vlachoudis/bCNC #1249: This is not a problem, do not know that bCNC allows you to read canned / repetitive gcode

MARIOBASZ, 2019-07-05:
I have just discovered that bCNC interprets G80, G81-G89 canned cycles, G98-G99, maybe you still need to control if you have an error (in principle it does not repeat cycle for G90 is what I have seen, although I do not know if it is important, I do not know if it is a usual practice, G83 I think is waiting to be enabled).
Good news, I share it in case other tamboco were aware.
Once again congratulations to Vlachoudis and to those who with their enthusiasm and talent contribute to this.
Greetings to all. Hugs, people.

Harvie, 2019-07-08:
Does this mean there is a bug in bCNC canned cycle implementation?

MARIOBASZ, 2019-07-08:
What I mean is that Grbl says he does not accept fixed cycles. bCNC does not clarify that it does accept fixed cycles, it would be good to advise this utility.
I have not reviewed in detail its operation, if there is any problem it seems minor

Harvie, 2019-07-09:
> bCNC does not clarify that it does accept fixed cycles

It DOES in both README and wiki:

https://github.com/vlachoudis/bCNC/blob/master/README.md
https://github.com/vlachoudis/bCNC/wiki/FileFormat

closed 2019-07-09 by Harvie