vlachoudis/bCNC #1250: Continuous position read out

redpiperbob3451, 2019-07-10:
Hi can you help please.
I am using a STM32 board and shield  after giving a command to the board bcnc continually scrolls the work and machine positions can this be stopped ?
[image]

Harvie, 2019-07-10:
Yes. You run grbl 0.9, currently the legacy GRBLv0 support in bCNC is not very good. So you have two options:

1.) Help us fix the code and improve the legacy GRBLv0 support
2.) Upgrade your STM32 to GRBL 1.1, which works fine

redpiperbob3451, 2019-07-10:
thanks for the reply 
1 I am just starting at thhis stuff so it may be some time before I am ready to help .2 I dont know how to do thisthanksBob

Harvie, 2019-07-10:
I've just entered "stm32 grbl 1.1" to google and this came out:

https://github.com/dungjk/grbl-stm32
http://www.allreadable.com/0b2eOCqG

redpiperbob3451, 2019-07-10:
Hi Tomas thanks for that I just did the same and I am reading up on this. Thanks for your help I will let you know how I get on. But as I say I am completely new to this.
Bob

 
 

https://github.com/dungjk/grbl-stm32
http://www.allreadable.com/0b2eOCqG

TheAntTeam, 2019-12-01:
Hello, 
we are using a modified version of grbl 0.9 for our "The Ant" machine, so we use the GRBL v0 controller. 
To solve this issue we commented in the GRBL0.py file the lines 46 and 47, and it seems to work.

Do you see any potential issue or side-effect in that?

Regards,
The Ant Team.

Harvie, 2019-12-01:
> Do you see any potential issue or side-effect in that?

I think the received line is probably not properly parsed. By commenting these two lines you just disable loging of that event... Not sure what are real side effects. Do you see proper coordinate readout (DRO) when using grbl 0.9?

BTW what are your reasons to use grbl 0.9? Can you tell us more about your ant machine?

carl1961, 2019-12-02:
Tomas Mudrunka  This it there videos, excellent machine, i built  one and
use it weekly to mill PCB's
Tomas Mudrunka  This it there

TheAntTeam, 2019-12-02:
Hi Harvie,
yes, we see see proper coordinate readout (DRO) and it seems everything works fine, in this sense.

What the Great Carl said is more than correct, and adding on some details, we ported grbl 0.9 J to a NUCLEO-F401RE board, it has a 32 bit STM32 processor and debugger and arduino shield compatibility. 
We used this board to have more flexibility in our development (e.g. we added some parameters for the spindle to drive an ESC).
The repository with everything about the machine and the firmware is here:

[Repository_THE_ANT](https://bitbucket.org/compactpcbmaker/cpcbm/src/master/)

Harvie, 2019-12-02:
> yes, we see see proper coordinate readout (DRO) and it seems everything works fine, in this sense.

OK! Good to hear that. But i'd still like to figure out why the message keeps falling into the log. There's probably some minor parsing problem which causes message to be parsed properly, while still throws an error.

TheAntTeam, 2020-02-08:
Hi, 
we finally had the time to look a bit into this issue, and we saw that in the following if, the part that fails is the first condition:

`if self.master.sio_wait and not cline and pat.group(1) not in ("Run", "Jog", "Hold"):`

Looking into "Sender.py", it seems that the "sio_wait" is not set to True for some messages, in particular for the serial polling, so for those messages that condition fails, and the else in the GRBL0 is always executed.
In the GRBL1 and SMOOTHIE controllers there is a similar if block in the "parseBracketAngle" function, but the else does not exist, and then it's not executed.
Hope this helps!

Harvie, 2021-02-23:
Can you please check latest pip release? This should be fixed by now...