vlachoudis/bCNC #1253: Scripting Unable to execute python command

excury, 2019-07-15:
Hi,

I try to use python inside G-code. The python print work. It execute it at the beginning and at that point in the G-code. 
The problem is the python script test3.py that do not execute. I use the % in front of the line. 

[image]

How to make the python script work?
All the help would be really appreciated.

Harvie, 2019-07-16:
I think that `run test3.py` is not a python command. See, i can't even run it without the bCNC:

[image]

excury, 2019-07-16:
Hi Harvie,

In that case, what is the proper command to the python script in BCNC?

Thank you very much in advance,

Harvie, 2019-07-16:
It's in the wiki: https://github.com/vlachoudis/bCNC/wiki/Scripting

You might use os.system() to launch any external command:

`%import os; os.system("python yourscript.py")`

closed 2019-07-16 by Harvie

excury, 2019-07-16:
Where does the python file need to be save because I have this error message?
FileNotFoundError: [Errno 2] No such file or directory: test3.py

I tried to put it in the same directory as the G-code file and in the same directory as the __main__.py bCNC file. I obtain the same error message for both.

Harvie, 2019-07-16:
just use full path then.

If you use os.system() it's same as if you've entered that command in linux terminal. you can add your command to $PATH or check in which directory is your bCNC running by `%import os; os.system("pwd")` just like you would use `pwd` in bash. Problem is that pwd might be different every time you launch bCNC, so better to use wrapper in $PATH or at least full path to script.

Sorry, but if you want to program in python, you will have to learn it first :-)
This is bit out-of-scope for bCNC project, but there are lots of online resources https://docs.python.org/3/library/os.html#os.system