vlachoudis/bCNC #1254: sketch plugin doesn't output anything - IndexError: image index out of range

zmaile, 2019-07-16:
The sketch plugin gives the following on any image and settings combination I have tried:
```
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/bCNC-git/Utils.py", line 460, in __call__
    return self.func(*args)
  File "/usr/lib/python2.7/site-packages/bCNC-git/ToolsPage.py", line 1746, in execute
    self.tools.getActive().execute(self.app)
  File "/usr/lib/python2.7/site-packages/bCNC-git/plugins/sketch.py", line 334, in execute
    self.fadePixel(x, y, pix, fading, repetition) #adjustbrightness int the bright map
  File "/usr/lib/python2.7/site-packages/bCNC-git/plugins/sketch.py", line 161, in fadePixel
    pix[x-3,y+3] +=1*fad
IndexError: image index out of range
```
The line

> pix[x-3,y+3] +=1*fad

does change with different settings/images, but always contains a +-3, which seems to cause the index out of range error.

The error occurs in the fadePixel function, however looking through bCNC/plugins/sketch.py, it seems that the findInRange function is the culprit, with xmin, xmax, ymin, ymax values being min/max of 2, not 3. findInRange is called before fadePixel, and sets the X/Y coordinates that causes the index error.

I haven't tested this fix, so i'm not sure if there are any other repercussions to changing the values from 2 to 3.

Harvie, 2019-07-16:
@sfinexer any ideas on this?

zmaile, 2019-07-17:
I tried my hand at a fix and managed to get it working on my machine.

[sketch.diff.txt](https://github.com/vlachoudis/bCNC/files/3400017/sketch.diff.txt)

I've never really contributed to a project before, nor am I a dev, so i wouldn't suggest implementing this patch without checking it doesn't bork something. But I hope it at least helps give an idea of the culprit code.