vlachoudis/bCNC #1255: Homing($H)ignore software limit is disable($20=0) only for X axis

Kagyroy, 2019-07-21:
Homing($H)ignore software limit is disable($20=0) only for X axis.

First, I found bcnc changes the origin xy axis followed by sofware limit distance($130~$132) after Homing when I change the direction of xy axis($3).

I wanted to avoid this problem because it cause Machine coodinates (-x sofwere limit, -y software limit, -z software limit).

Harvie, 2019-07-22:
I don't really understand what is your problem. I've never used homing, can you please explain?
Do you want to disable software limits? I think it can be done in GRBL..

Kagyroy, 2019-07-22:
Sorry for my poor explanation..

Problem 1:
When I enabe software limit, it works while jog.
But it doesn't work on Homing proccess(only X axis, Y )

closed 2019-07-22 by Kagyroy

Harvie, 2019-07-22:
This seems to be GRBL related problem, does it happen only with bCNC or also with other GRBL UIs?