vlachoudis/bCNC #1256: Dialog Box in Button script?

BernardG, 2019-07-26:
As many others, I would like to write a script on a button to probe on X, Y and Z. The thing is, as you know, is that, if the script is pure GCode, I will need to use one button per tool diameter. (not even talking about choosing which corner to probe)

So, I wonder if there is a way to use some python code to ask the user his tool diameter, and use the variable value in GCode?

I know about the possibility to write plugins, but that does not look (to me) as a plugin job, should be juste used on one button, if at all possible.

Thanks for a great piece of software!

Harvie, 2019-07-26:
Never tried this, but from my understanding, this should be possible according to https://github.com/vlachoudis/bCNC/wiki/Scripting

First, you have to asign variable like this:  `%_yourvariable=123` then you can use it in g-code like eg.: `G0 X10 Y#yourvariable`. You can replace the asignment with any arbitrary python line, so you should be able to make python popup that asks for a value to asign to _yourvariable.

Harvie, 2019-07-26:
So basicaly i've found workaround for you and mentioned in wiki. Code goes like this:

```
%_myval=raw_input("enter value: "); print("entered value: "+_myval);
G0 X10 Y[_myval]
```

[image]

In future we can upgrade this to provide popup, but this should be good enough to get the work done for now...

BernardG, 2019-07-26:
Yes, that will be good enough for now. Thank you so much! Sure a popup would be "more classical" and look nicer, but this will get the job done, which is what really matters.

Harvie, 2019-07-26:
I've just tested and it also works in buttons:
[image]

What OS do you use? It's important to open bCNC from commandline.

BernardG, 2019-07-26:
I have been trying with Ubuntu Linux, not yet from the command line in Windows 10.
Where do you enter asked value? On bCNC command?

Harvie, 2019-07-26:
Enter the value to the terminal and hit enter.

BernardG, 2019-07-26:
OK, got it. It's just such an ugly workaround that I did not even think that was the way to go! :-)
Now, what would be the proper way yo call a python popup instead?

Harvie, 2019-07-26:
@BernardG don't worry, i think i might roll some new code for this today... i am very close!

Harvie, 2019-07-26:
I got it working, will upload in minute:

[image]

BernardG, 2019-07-26:
I do appreciate your work, but in my tests (with the terminal), I get an error from python "cannot concatenate str into int (or float), as I need to be able to divide by 2 the diameter. Have been trying to do _myval = float(myval), but as I say, it gives an error. Any idea?

BernardG, 2019-07-26:
just tried your suggestion, I get "app is not defined".

Harvie, 2019-07-26:
> just tried your suggestion, I get "app is not defined".

i said i will upload it in a minute, which i just did. Try with latest git or pypi version of bCNC now. just tried and it works :-)

BernardG, 2019-07-26:
So I need to update from the repository and test? Waow, sound great! I'll do that tomorrow.

Harvie, 2019-07-26:
> So I need to update from the repository and test? Waow, sound great! I'll do that tomorrow.

do you have bCNC installed using pip? In such case it's just matter of doing something like 

`pip2 install --upgrade bCNC`

BernardG, 2019-07-26:
> > just tried your suggestion, I get "app is not defined".
> 
> i said i will upload it in a minute, which i just did. Try with latest git or pypi version of bCNC now. just tried and it works :-)

yes, sorry, did not understand at first you were making a modification to the code.
I believe yes, I installed it that way...Let me do this....

BernardG, 2019-07-26:
Thank you so much, this works perfectly. So now you can close this issue! :-)

closed 2019-07-27 by Harvie