vlachoudis/bCNC #1258: bCNC install fails on Mac

loonster, 2019-07-27:
I get this error on a Mac OS Mojave:  Would appreciate any help!!

~ $: pip2 install --upgrade bCNC
DEPRECATION (message)
Collecting bCNC
  Using cached https://files.pythonhosted.org/packages/9d/24/bc3e26eb2ec232dfc591b7ecc6fb6d486d920320412b75e111509165fe68/bCNC-0.9.14.60.tar.gz
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-install-L_Nygj/bCNC/setup.py'"'"'; __file__='"'"'/private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-install-L_Nygj/bCNC/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base pip-egg-info
         cwd: /private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-install-L_Nygj/bCNC/
    Complete output (6 lines):
    /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)
    /System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'long_description_content_type'
      warnings.warn(msg)
    Running bCNC setup...
    error in bCNC setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers

Harvie, 2019-07-28:
Can you try this? `pip2 install --upgrade pip`

loonster, 2019-07-28:
Thanks for the response.  Yes, that was the first thing tried.  The system:  “Requirement already up-to-date: pip in /Library/Python/2.7/site-packages (19.2.1),

Thinking that the setuptools, already installed, needed to be upgraded, I issued the command
'python -m pip install —upgrade pip setuptools wheel’ but this command failed:

ERROR: Could not install packages due to an EnvironmentError: [('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', '/private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-uninstall-0BZMb0/__init__.py', "[Errno 1] Operation not permitted: '/private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-uninstall-0BZMb0/__init__.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', '/private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-uninstall-0BZMb0/__init__.pyc', "[Errno 1] Operation not permitted: '/private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-uninstall-0BZMb0/__init__.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', '/private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-uninstall-0BZMb0/markers.py', "[Errno 1] Operation not permitted: '/private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-uninstall-0BZMb0/markers.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', '/private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-uninstall-0BZMb0/markers.pyc', "[Errno 1] Operation not permitted: '/private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-uninstall-0BZMb0/markers.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', '/private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-uninstall-0BZMb0', "[Errno 1] Operation not permitted: '/private/var/folders/60/8_j1y6rx36597cb_gbxj7ylh0000gn/T/pip-uninstall-0BZMb0'")]

I then issued the last command with sudo, and that seemed to have worked…but when I issued the usual python -m pip install —upgrade bCNC, the result is the same error as in my original post…

Harvie, 2019-07-28:
Have you tried installing bCNC with sudo?

loonster, 2019-07-28:
Yes; same error.

Harvie, 2019-07-28:
Please provide output of `python2 --version` and `pip2 freeze` commands.

Harvie, 2019-07-28:
Also note that you should always run pip install as root.
If you want to do pip install under different user, then you should always do it like this:

`pip2 install --user --upgrade bCNC`, that way bCNC will be installed to user home instead of system-wide directory. which might be usefull, because it does no require root privileges. But it will be only available for that one user.

Also you might experiment with `--no-cache-dir` flag.

loonster, 2019-07-28:
Here they are:

loonster, 2019-07-28:
sudo python -m pip install —upgrade bCNC —no-cache-dir

…produces the same old error.  

also note my system says “No module named pip2"

Harvie, 2019-07-28:
> Here they are:

I guess the email didn't get through properly

loonster, 2019-07-28:
OK:

~ $: sudo python2 --version
Password:
sudo: python2: command not found
~ $: sudo python --version
Python 2.7.10

****************************************

~ $: sudo pip2 freeze
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
altgraph==0.10.2
bdist-mpkg==0.5.0
bonjour-py==0.3
macholib==1.5.1
matplotlib==1.3.1
modulegraph==0.10.4
numpy==1.8.0rc1
Pillow==6.1.0
py2app==0.7.3
pyobjc-core==2.5.1
pyobjc-framework-Accounts==2.5.1
pyobjc-framework-AddressBook==2.5.1
pyobjc-framework-AppleScriptKit==2.5.1
pyobjc-framework-AppleScriptObjC==2.5.1
pyobjc-framework-Automator==2.5.1
pyobjc-framework-CFNetwork==2.5.1
pyobjc-framework-Cocoa==2.5.1
pyobjc-framework-Collaboration==2.5.1
pyobjc-framework-CoreData==2.5.1
pyobjc-framework-CoreLocation==2.5.1
pyobjc-framework-CoreText==2.5.1
pyobjc-framework-DictionaryServices==2.5.1
pyobjc-framework-EventKit==2.5.1
pyobjc-framework-ExceptionHandling==2.5.1
pyobjc-framework-FSEvents==2.5.1
pyobjc-framework-InputMethodKit==2.5.1
pyobjc-framework-InstallerPlugins==2.5.1
pyobjc-framework-InstantMessage==2.5.1
pyobjc-framework-LatentSemanticMapping==2.5.1
pyobjc-framework-LaunchServices==2.5.1
pyobjc-framework-Message==2.5.1
pyobjc-framework-OpenDirectory==2.5.1
pyobjc-framework-PreferencePanes==2.5.1
pyobjc-framework-PubSub==2.5.1
pyobjc-framework-QTKit==2.5.1
pyobjc-framework-Quartz==2.5.1
pyobjc-framework-ScreenSaver==2.5.1
pyobjc-framework-ScriptingBridge==2.5.1
pyobjc-framework-SearchKit==2.5.1
pyobjc-framework-ServiceManagement==2.5.1
pyobjc-framework-Social==2.5.1
pyobjc-framework-SyncServices==2.5.1
pyobjc-framework-SystemConfiguration==2.5.1
pyobjc-framework-WebKit==2.5.1
pyOpenSSL==0.13.1
pyparsing==2.0.1
pyserial==3.4
python-dateutil==1.5
pytz==2013.7
scipy==0.13.0b1
six==1.4.1
xattr==0.6.4
zope.interface==4.1.1

Harvie, 2019-07-28:
I have Python 2.7.16, do you think it will be possible for you to upgrade?

loonster, 2019-07-28:
Yes.  Will look into it now...

loonster, 2019-07-28:
It appears that upgrading the Apple-installed python is a very bad idea, since so many dependencies on it exist.  Perhaps adding a newer version via Homebrew is the answer.  Is there a way to specify, say, version 2.7.16 or an even higher version for such an installation?

Harvie, 2019-07-28:
Can't you just download and install it from here?
https://www.python.org/downloads/release/python-2716/

loonster, 2019-07-28:
I’ll try it…am concerned with mucking up the existing install and the other software on the computer that is dependent upon the Apple-supplied version….but what the hey….the machine is old and will probably not be supported into the future and I can always wipe it clean and install linux.

Harvie, 2019-07-28:
> I can always wipe it clean and install linux.

Haha. That's always good idea, IIRC even Linus Torvalds used to use Mac reinstalled with Linux in the past :-)
However before you do so, i would love to figure out if bCNC still actualy works on Mac and what can be improved to satisfy basic needs of most MacOS users. I don't have Mac and we don't have anyone testing and reporting issues with bCNC on Mac, so i am not really sure about current status of MacOS support.

loonster, 2019-07-28:
Success!  I downloaded 2.7.16 and bCNC opens just fine.  After doing some errands now, I’ll test it out on my Shapeoko3 later today.  I’ll let you know how it goes.  Much appreciate your help.   And I’d be happy to do testing or whatever; add me to the loop, so to speak.  

I do have a computer science degree, but haven’t programmed much in python…and being retired, I’ve got the time.  A question:  is bCNC going to be ported to python 3+???  

Tim Burton
tburton@acm.org

“We don’t see things as they are, we see things as we are.”  — Anais Nin

Harvie, 2019-07-28:
> Success!  I downloaded 2.7.16 and bCNC opens just fine. 

Glad to hear that. Please report if you encounter some issues.

> A question:  is bCNC going to be ported to python 3+???

Yes, there is already lot of discussion on this, you can find it here on github. However the thing is that i wasn't able to do it. I've tried several times, but it would probably need much more effort or experience that i have. I've run the bCNC code through some automated scripts, which should migrate it to python3 and then tried to fix the rest. Didn't managed this last step. At least current bCNC codebase hopefully follows some best practices for future porting to python3. As i was trying to migrate to python3, there were some python2 compatible modifications needed to be done and i've left them in python2 version, so it might be bit easier to migrate later. But unfortunately that might broken compatibility with older 2.x versions. But 2.7.x seems to work reasonably well..

ccomincini, 2019-09-28:
I had same troubles with OSX and, at last of my troubles, I started back from scratch and solved like this:

- [ ] first of all I've installed TK following instruction on the official page at [https://tkdocs.com/tutorial/install.html#installmac](https://tkdocs.com/tutorial/install.html#installmac)  _look for the python section_. Thish tutorial wll guide you in installing also python from source, configuring it with the correct parameters to run Tkinter.
_an hint for the beginners : Note thet the page give you teh "_configure_" command after that you have to run "_make_" and "_make install_"_

After that, you can install bCNC using

- [ ] sudo pip install --upgrade pip --user
- [ ] sudo pip install --upgrade setuptools --user
- [ ] sudo pip install --upgrade bCNC --user
_remember to add the --user  parameter for the command pip._

I really guess if "sudo " is  mandatroy, but it worked for me. 
Hope this can be useful for people using mac and not being a developer.