vlachoudis/bCNC #1259: Remove OLD Profile Paths and inside and outside recognision

dylanpil, 2019-07-29:
I LUV bcnc and all the developments that keep growing!!
I'm not too sure if i am putting this request in the right area.

When creating profiles, a great feature would be a check box that when checked would delete the original paths. Presently i go back and delete each path individually.
A second great feature would be a profile option that would profile inside shapes as inside offsets and outside profiles as outsides, presently one has to do these as 2 separate operations.
A third nice option  would be to add a check box to make it either Std or reverse cutting (ie for wood verses plastic cutting).

Keep up the amazing developments to this fantastic program!

Harvie, 2019-12-01:
Yes. I am aware of this. Biggest problem is that the behaviour is not consistent across the tools.
Maybe we can have some central checkbox that will affect all the tools at once, so people will be able to choose what behaviour they prefer... Or maybe clicking on the button with CTRL pressed should cause it to delete the original block, otherwise it should duplicate or something. IDK.

MARIOBASZ, 2019-12-01:
It is a good option to keep / delete the original block, although I do not prefer it: on the contrary, I have modified it on my computer and the new blocks have the source name and comment of the tool used and if I have inserted offset:
block [bit6.00 off-0.2]
In case I use after a while.
I keep the original blocks off and at the end of the list