vlachoudis/bCNC #1261: Setting 0,0 Point in Preview

barnold96, 2019-08-06:
I use the center of all my projects as the 0,0 point. When I import a project into bCNC, it assumes 0,0 is the back right point and centers my code there. 

In the preview, there is a square that appears to represent the work area. An imported project centers over the top, right corner of that square. My imported project shows the correct 0,0 starting point and I can move it to the center of the preview window.  The project engraves correctly.

Is there a way to tell bCNC that 0,0 is the center of a project?

[image]

Harvie, 2019-08-06:
What? 0,0 is where you put it in the CAD software, bCNC only loads it as it is.
However you can change it by mouse or to one of these predefined points:

[image]

closed 2019-08-06 by Harvie

barnold96, 2019-08-06:
My origin is exactly where I want it to be: the center of a project.  I don't have to change anything in any of my drawings or software to do what I want to do.

My question is simple enough: Is there a way in bCNC to tell it my workspace has 0,0 in the center?

Harvie, 2019-08-06:
I don't understand your problem. You say you already have origin in center where you want it to be, so what do you need to change?

barnold96, 2019-08-06:
In bCNC, 0,0 defaults to top right.  Is there a place in its configuration where one can define the size of the workspace on a machine and tell bCNC the origin point is the center of that workspace?

Harvie, 2019-08-06:
> In bCNC, 0,0 defaults to top right

No it does not. 0,0 defaults to where you put it. Can you share your file, so i can check what the problem is?

> Is there a place in its configuration where one can define the size of the workspace on a machine and tell bCNC the origin point is the center of that workspace?

It does not work like this. You might need auto-homing setup for this, then you can configure WCS with 0,0 in center. Otherwise you will need to place some mark in middle of your workpiece, jog to it and hit zero every time you (re)start your machine.

Harvie, 2019-08-06:
Oh! Now i understand what you needed:

[image]

I had this "frame" thingy disabled. I never use it, i don't think it's needed. I managed to swap the origin by entering negative numbers. I guess you should enter half of your workspace or disable it completely if you want it to be in center.

However this can be easy fix in code to add checkbox for workspace in center or something. It's very low priority right now, because there's not a big practical use for this. However if you manage to write the code, i will merge it.

barnold96, 2019-08-06:
Ah-ha!  I set all three 'Travel' numbers to 0(zero).  Now, when I import a drawing, it fills the preview window with no further manipulation, regardless of the size of the drawing.

Thanks!

Harvie, 2019-08-06:
It worked for me exactly the same all the time, all i needed to do was unclick the little orange square button at the top of the canvas... No need to change the coordinates if you just wanted to hide it...

barnold96, 2019-08-06:
OK, thanks.  I see that works, too.