vlachoudis/bCNC #1262: help! new to CNC milling

lawrenraggio, 2019-08-10:
Hi, 
I'm new to this, am an artist and bought a CNC machine to mill text onto wood charcuterie trays I make and sell. 
I have a macbook pro (no cd drive) and the machine came with a cd I'm assuming has everything I need. 
I am running parallels, and have been playing around with programs such as Arduino, ABViewer, Textmilling4, Camotics, Inkscape and Meshcam. 
My CNC machine is SainSmart Genmitsu CNC 3018 Router Kit GRBL Control 3 Axis. I put together without directions (as they were terrible).
I cannot seem to get the machine to work and am at a loss. Am close to returning entire thing. 
If anyone can point me in the right direction, I'm all ears! 
Signed, 
Frustrated Artist, Lorii

DeanRM, 2019-08-10:
This is a forum for bCNC software.  Your best bet for help is join a CNC beginners group on facebook.  The couple I have joined are aways willing to answer questions.

Harvie, 2019-08-10:
As long as it's really GRBL board, you should be able to connect using bCNC to it.
If that does not work and there's a chip with "atmega" written on the board, you might be able to flash latest version GRBL firmware to it using arduino software and instructions on https://github.com/gnea/grbl/wiki
then it should work with bCNC.

closed 2019-08-10 by Harvie