vlachoudis/bCNC #1263: 22,000 line limit?

woodworkerbob, 2019-08-10:
I'm engraving a large piece. I created two different gcode files from Aspire, as I have done for 5 years, and both files fail at line 22,001. Failure mode. Complete project does not finish drawing in the graphics window. Bringing up the editor, I followed the gcode to the line where it no longer drew in the graphics window.  

Used both the newest and an old downloaded version of bCNC.
How did I screw up?

Robert
Aspire going to Shapeoko3 with GRBL 1.1

Harvie, 2019-08-10:
> Complete project does not finish drawing in the graphics window. 

bCNC currently uses very slow graphics library, so the rendering is time limited, so it does not hang for too long on big files. You can set the limit here:

[image]

It should not affect the milling.

If your milling fails after 22k lines, there is probably something else wrong. Usualy it boils down to one of these:

1.) g-code error - some invalid g-code is found in file
2.) hardware error - usb serial chip on your cnc controller board corrupts the g-code, or there's some kind of EMI problem
3.) actual bCNC bug - nobody was able to prove that there's such a bug in current git master version, maybe you will be the first

tklus, 2019-08-10:
I use BCNC for running my CNC router/laser machine. I regularly run programs that are between 40,000 and 90,000 lines of code. I have a 3W laser attached that I use for engraving.

I use laser etch from Jtech for generating my gcode.

inspect your g-code and see what is there at line 22K

Hope this helps,
Tim

woodworkerbob, 2019-08-10:
It was the time out parameter of the drawing-graphics window. The timeout
was set to 5. I changed it to 20 and all is good.

Thanks for the help.

Robert

yrabbit, 2019-08-10:
Don’t worry about the limit --- I successfully use files with half a million lines
rabbit@fly ~/tmp% wc ~/src/ps2kbd/out/BCu-fine.gcode
  590955 2307402 16533877 /home/rabbit/src/ps2kbd/out/BCu-fine.gcode

woodworkerbob, 2019-08-10:
Just changed rendering time limit factor to 20. That solved the graphic issue.  I was a bit panicked as I've used bCNC for years and on much larger files and didn't remember this issue.  bCNC is my go-to tool for controlling my Shapeoko 3.

Really love the auto leveling feature for small sized engravings on wooden surfaces. I get consistent even lettering.

Thanks

Harvie, 2019-08-10:
[OFFTOPIC] @yrabbit is that a PCB g-code? what do you use to generate it? I've done few experiments with KiCad+flatcam workflow, but it seems to me that flatcam is not really optimized to work with KiCad gerbers... Resulting g-code needed lots of tuning in bCNC. (BTW this got me thinking of adding gerber import to bCNC :-)

closed 2019-08-10 by Harvie

yrabbit, 2019-08-11:
My chain is simple: KiCad -> fcad_pcb -> FreeCAD -> bCNC
No export from KiCad is performed, fcad_pcb(https://github.com/realthunder/fcad_pcb) works directly with its files, just as I don't edit anything in g-code or inside the bCNC.
And now it’s clear where the catch is --- you should be really good at working with #FreeCAD :), moreover, if you want to use adaptive cleaning of areas like I do, then you should contact me (there are a few subtleties).
[image]
[image]
[image]

MARIOBASZ, 2019-08-11:
yrabbit2, what do you mean by adaptive cleaning? I don't know if it's a similar approach I'm working on: using two tools, for example, 0.7 mm to route the tracks and 0.1 mm in the areas where the wide tool cannot enter due to an overlap problem. (I'm bad at welding, too good to generate short circuits ... separating tracks helps me)
I have been advancing in this complement, although I still lack knowledge to accelerate the processes: it takes approximately 5 minutes of processing (before it took about 45 minutes)
Perhaps the solution has to do with multi-thread (I don't know how it is implemented) or with python gis?
Importing gerbers directly is a great idea. Importing from brd files would be good (I think chilipeppr does something like that)
The driller approach with excellion is useful. Another improvement would be to use a bit for all holes, generating helical descents with corresponding radius.

yrabbit, 2019-08-11:
hmm, I clean the areas and you do isolation routing. My way is much slower and there is only one tool with a difficult path :)
[image]

Harvie, 2019-08-11:
Unfortunately i am unable to get FreeCAD running on ArchLinux for more than a year, there is always some bug. Even the AppImage version crashes.

MARIOBASZ, 2019-08-11:
I did not know the existence of fcad_pcb (is it responsible for exporting to gcode?), nor have I used free cad: does it generate adaptive cleaning avoiding overlapping adjacent tracks? Adaptive cleaning can be done from bcnc with trochoid 3d, it still seems almost as attractive as Brigite Bardot in 1970 to see my machine milling aluminum in passes of 8mm deep without suffering.
It remains to be able not to enter adjacent tracks, if I can complete pcb with two strawberries, that solution will also be used for adaptive cleaning, since the path matches.
From Eagle pcb gcode is responsible for exporting with successive passes for a non-trochoidal adaptive cleaning, but it is too long.
As I am also trying to finish the design of a low-cost ATC, if my idea is successful, using two strawberries will be very useful

MARIOBASZ, 2019-08-11:
I must look at free cad: Maybe the best and fastest solution is to use free cad, and suggest if they implement adaptive with two bits.

MARIOBASZ, 2019-08-11:
Adaptive fusion respects not entering adjacent shapes?

yrabbit, 2019-08-11:
No, fcad_pcb does not export g-code, it reads KiCad files and creates 3d models of tracks, pads, holes, boards, etc. Then you can use FreCAD to create tracks for milling and drilling and generating g-code.

Unfortunately, I can’t tell anything about FreeCAD on Linux --- I use DragonFlyBSD as main workstation.

About adaptive operation in FreeCAD:
https://forum.freecadweb.org/viewtopic.php?f=15&t=30127