vlachoudis/bCNC #1264: Open KiCad gerbers X2 and possibly excellon files.

Harvie, 2019-08-11:
I would like to add support for opening gerber files, because flatcam does not really work with KiCad gerbers for me and i have to fix the g-code in bCNC anyway.

Opening excellon files might be option as well. Right now it's implemented as part of driller plugin #809, which is not really nice approach. I think we should be able to open it through open dialog and convert it to g-code like we do for all other file formats.

On the other hand excellon file format is deprecated as new version of gerber format called "gerber X2" also does drilling.

```bash
cat baseplate-F_Cu.gbr | grep -v % | grep X | sed -e 's/[*X]//g;s/Y-/ /g;s/\(.*\)D01/G1 X\1/g;s/\(.*\)D02/G0 X\1/g;s/\(.*\)D03/\(D03 \1\)/g;s/ / Y/g;' | LANG=C sed -e  's/\(......\) /.\1 /g;  s/\(......\)$/.\1/g;    s/YX/X/g' > test.ngc
```

MARIOBASZ, 2019-08-30:
I find this interesting, I will read it.
I don't know about the existence of Gerber X2.
Reading grbl directly can be an option?
The piercing plugin is not bad either.
Modifying so that with a single bit it generates the different diameters would be efficient.

corpix, 2025-05-06:
I was searching for a solution and found this script https://github.com/jes/drl2gcode/blob/master/drl2gcode.py
It generates separate gcode file for distinct hole size. Works perfectly for me!