vlachoudis/bCNC #1265: bCNC fitted to 5 inch LCD Screen

lyatesbme, 2019-08-12:
I am using a 5" LCD screen for bCNC to control a 3018 CNC unit. It loads up well using the availble image that includes the bCNC but I can't seem to fit it to the screen and there is functions off screen.
I have seen an old thread about this and the term used was to "undecorate" but can't seem to find any later references. Can you help and/or advise.
Many thanks

Harvie, 2019-08-26:
Duplicate of #1033 , please feel free to continue there. Also mention resolution of your LCD rather than it's physical size.

closed 2019-08-26 by Harvie

lyatesbme, 2019-08-27:
Many thanks for your repsonse

Larry Yates
Manager Biomedical Engineering and Technology.
President S.M.B.E. N.S.W.
B. Adult Ed, Diploma Electr. Eng, Grad Cert Public Sector Management
02 68097936     0429660091
larry.yates@health.nsw.gov.au

This message is intended for the addressee named and may contain confidential information. If you are not the intended recipient, please delete it and notify the sender.

Views expressed in this message are those of the individual sender, and are not necessarily the views of NSW Health or any of its entities.