vlachoudis/bCNC #1266: bCNC Pause @ 9999

barnold96, 2019-08-21:
First, some background:  I run my laser engraver via RPi3 for wireless connectivity, with bCNC running on it and accessed via VNC.  I haven't had any problem running a variety of projects until a couple days ago when I started experimenting with photographs.

I'm trying out Lightburn to process a digital photo and create gcode.  I used the Lb default of 0.10mm line with for a photo I reduced to 2.75" wide by 3.5" high.  A gcode file length of 325,000 lines was generated.  When I run that gcode on bCNC, 9999 lines are shown in the progress bar at the bottom of the window.  The engraving seems to run OK, timing itself based on the 9999 lines, then it pauses.  After a few seconds, the additional lines are shown in the progress bar and the run finishes.  I ran it three times with the same result.

Next, I wanted to see what happened if I doubled the resolution to 0.050mm line width. The file jumped to 650000+ lines. I loaded it and the progress bar again showed 9999 lines initially. When it got to line 9999, it paused and shut down as if finished.  bCNC never resumed the engraving.

When I check the gcode file with Notepad++, it shows nothing unusual surrounding line 9999 on either version.  I have run several other projects that are over 9999 lines without the same results.

Is there some kind of limitation I'm missing?

Harvie, 2019-08-26:
I am not sure, but i think there was some kind of trick that bCNC shows 9999 lines as initial estimate of total line count even when there's actually more, because it takes some time to preprocess all these lines. So that's not accurate. This is currently "normal" and should not affect your operation.

Maybe there is some issue on different line? Do you see what was the last line actually processed?

Harvie, 2019-08-26:
Send me your g-code file and what OS you use. Upgrade to latest git master or pip version and try again.

trihwangyudi, 2019-08-29:
Hey i have the same issue, but after i change bCNC to new it's work.
I change from bCNC with windows OS to bCNC with linux mint

barnold96, 2019-09-19:
Is anyone looking at this?

Harvie, 2019-12-01:
Have you tried latest git master version of bCNC?

trihwangyudi, 2019-12-01:
hey i found somethings wrong with arduino chinese ch340, are you using this type board?? just change it with original one or others type higher than ch340

alessnilsen, 2020-09-06:
Hello! I have this same issue with my laser printer. I usually work with very large gcode files containing raster images, when I load one of those gcodes into bCNC and press Play, the print goes very slow and erratic until it hits the 9999 lines sent, then it pauses briefly and resumes printing, but this time smoothly as it should be. The photo printed ends with the beginning screwed up because of the erratic behavior.

My printer runs grbl 1.1f on a STM32 blue pill board.
My bCNC version is 0.9.14-dev (downloaded 06/09/2020).
My computer runs Ubuntu 18.04 x64.

I can't find anything that cites the 9999 behavior. Attached is two screenshots of the problem and the gcode I'm using to reproduce it.

[t_0004_Escala_de_Cinza_256_.gcode.gz](https://github.com/vlachoudis/bCNC/files/5180019/t_0004_Escala_de_Cinza_256_.gcode.gz)
[image]
[image]

Harvie, 2021-02-23:
@barnold96 does your issues still persist with latest bCNC (pip release) and latest grbl 1.1?