vlachoudis/bCNC #1267: Raspberry Pi/ bCNC for X Carve - autostart program?

metb40, 2019-08-24:


metb40, 2019-08-24:
I would like the program to auto-start/boot-up directly to bCNC without having to launch it from the desktop { python -m bCNC }. Most of the ways I have seen posted for other programs to auto-start were a bit dated and I have had no luck getting them to work with bCNC. Any suggestions?

Harvie, 2019-08-26:
This is unrelated to bCNC and you would be much better discussing this with support of desktop environment that you are currently using. I suppose that you want to start bCNC everytime your raspberry boots Linux. It goes like this:

1.) You need your system to be configured to automaticaly login on boot
2.) You need to add bCNC to autostart of your desktop environment

This can be done by putting `.desktop` file to `~/.config/autostart`. If you manage to do this, please describe the process on our wiki: https://github.com/vlachoudis/bCNC/wiki other users might appreciate it.  Thanks :)

closed 2019-08-26 by Harvie