vlachoudis/bCNC #1268: USER Buttons saved where?

BernardG, 2019-08-25:
After spending quite a bit of time writing a+-20 lines script/macro, and after closing/reopening bCNC, I saw the macro was not there anymore. The name was still there, but no text.
On 2 other buttons, it all went away, name and text!

So, where does bCNC save this, so that I could eventually get it back? (I am afraid to close bCNC, now....)

Harvie, 2019-08-26:
bCNC.ini in your home directory (depending on OS that you use). You can also open it here:

[image]

BernardG, 2019-08-26:
Thanks for your answer. But "something" does not look right:
Here is an extract of my user file:
[image]

And here is a part of what actually is on that button:
[image]

So, either it is in another directory, or.... I have no idea what's happening here!

Harvie, 2019-08-26:
Can you please copypaste here whole text of your script, so i can try it?

BernardG, 2019-08-27:
Sure! Works fine for me, on the lower left corner of a piece, as is. I test with aluminum paper, which why thickness is @ 0.1. Comments in French, sorry.... ;-)

`%_mydia=app.entry("Entrez le diametre");
%_demidia=float(_mydia)/2;
%_epZ = float(0.1);
%_epX = float(0.1);
%_epY = float(0.1);
%_correctZ = str(_epZ);
%_correctX = str(_demidia + _epX);
%_correctY = str(_demidia + _epY);
%msg Debut de Sondage
G92.2 (Suppression de toutes les valeurs d'offset existantes)
G38.2 Z-35 F100 (Sonde en Z de 45 mm maximum à 50 mm/min)
G92 Z[_correctZ] (store la valeur trouvée + 0.1 mm-essai feuille alu)
G91 G0 Z3 (remonte à 3mm rapidement)
G91 G0 X-25 (Recule rapidement en X de 50 mm)
G91 G0 Z-5 (descend à Z-5 rapidement, en préparation du sondage en X)
G38.2 X25 F100 (sonde en X de 50mm - On doit forcemment trouver, puisque c'est de là qu'on vient)
G92 X-[_correctX] (store la valeur trouvéé + correction)
G91 G0 X-3 (recule de 3mm en Z)
G91 G0 Y-25 (Recule rapidement en Y de 50 mm)
G91 G0 X25
G38.2 Y25 F100 (sonde en y de 50mm - On doit forcemment trouver, puisque c'est de là qu'on vient)
G92 Y-[_correctY] (store la valeur trouvée + correction)
G91 G0 Y-3 (recule de 3mm en Y)
G91 G0 Z20 (remonte de 10mm)
%wait
%msg Sondage Terminé
%app.setStatus("Vous pouvez cliquer sur le cercle dans Control pour vérifier")`

BernardG, 2019-08-27:
Sorry, I did not realize I closed this thread, but I see this comment will allow me to reopen!

Harvie, 2019-08-27:
Yes. When i tried to save this, it gave me following error:

```
Traceback (most recent call last):
  File "/usr/bin/bCNC", line 11, in <module>
    load_entry_point('bCNC==0.9.14.60', 'console_scripts', 'bCNC')()
  File "/usr/lib/python2.7/site-packages/bCNC/__main__.py", line 2680, in main
    Utils.saveConfiguration()
  File "/usr/lib/python2.7/site-packages/bCNC/Utils.py", line 187, in saveConfiguration
    config.write(f)
  File "/usr/lib/python2.7/ConfigParser.py", line 412, in write
    key = " = ".join((key, str(value).replace('\n', '\n\t')))
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe0' in position 351: ordinal not in range(128)

```

This means that (probably french) national characters caused this issue. I think it will work for you if you will only use basic alphabet.

Also it caused my whole bCNC.ini to be corrupted. So this needs to be fixed.
There are two ways of fixing:

1.) Remove problematic characters from code before saving. This is workaround, but might help.

2.) Port whole bCNC to python3, which (unlike python2) has proper unicode handling. There are more unicode related issues in bCNC right now, so that would help a lot (eg. special characters in filenames cause troubles!)

BernardG, 2019-08-27:
That's an easy workaround, thanks Harvie. I did not think about this a s a possible issue.

BernardG, 2019-08-28:
OK, so I did as suggested, removed all french accented letters, then closed bCNC. Upon re-opening it, the command part is blank! Totally erased! Thanks bCNC!
Two things: 
-1- if bCNC is unable to save a macro, it would be nice to say so, so that the user could save the text somewhere else
-2- Is there some kind of max number of lines, or other limit, to what can be put in a macro?

BernardG, 2019-08-28:
It's correct now, bCNC did save the button macro properly. Maybe I had forgotten one accented letter, but it did truncate the whole bCNC.ini file..... (in my case, .bCNC)

Harvie, 2019-08-28:
Yes, python2 causes lots of annoying bugs related to unicode. Also python2 will not be officialy supported/maintained after january 2020, so i would love to figure this out. But i am not sure how to port bCNC, i've never managed to get it to work on python3.

BernardG, 2019-08-28:
I don't know Python enough to be able to help! Sorry.

Harvie, 2019-11-28:
It should be fixed in latest git master (not in pip release yet). Can you please try?

BernardG, 2019-11-28:
Hi Harvie, you mean, by downloading the source, right?

Harvie, 2019-11-28:
You can either download it here: https://github.com/vlachoudis/bCNC/archive/master.zip
Or use `git clone https://github.com/vlachoudis/bCNC.git` command (which is better, because it will enable you to use git for updating to latest development code if i change somethin, so you just do git pull and don't have to re-download whole .zip)

Harvie, 2019-12-01:
Please let me know if this is fixed for you. I've tried with czech alphabet and it seemed ok in both python2 and python3.

BernardG, 2019-12-01:
Give me some time to test. I'll do that early next week.

sttts, 2020-08-31:
This is still broken if you use umlauts, i.e. unicode characters. Keep backups of the config.

Miniclubbin, 2020-12-08:
This is still an issue, configured buttons will not persist after exiting. Raspberry pi 3.

Ogurezzz, 2020-12-22:
Update. Same. Sometimes configuration doesn't saves.
I made some buttons. Added endmills, then I confugured other stuff and only last change was saved on exit.
I think you need to add Save Config button for manual save everything.

swarfer, 2021-01-07:
Hi
Saving a macro fails if the macro contains , or any line beginning %
%msg "print this"

Has the macro syntax changed for P3?

python 3.9.1 64bit
bCNC 0.9.14-dev

Traceback (most recent call last):
  File "c:\python3\lib\site-packages\bCNC\Utils.py", line 526, in __call__
    return self.func(*args)
  File "c:\python3\lib\site-packages\bCNC\Utils.py", line 860, in ok
    config.set("Buttons", "command.%d"%(n), self.command.get("1.0",END).strip())
  File "c:\python3\lib\configparser.py", line 1201, in set
    super().set(section, option, value)
  File "c:\python3\lib\configparser.py", line 894, in set
    value = self._interpolation.before_set(self, section, option,
  File "c:\python3\lib\configparser.py", line 402, in before_set
    raise ValueError("invalid interpolation syntax in %r at "
ValueError: invalid interpolation syntax in 'G90 G0 X0\n%msg "write this"' at position 10
(<class 'NameError'>, NameError("name 'self' is not defined"), <traceback object at 0x000002157F7DD580>)

Harvie, 2021-02-23:
> This is still broken if you use umlauts, i.e. unicode characters. Keep backups of the config.

Seems to work for me with latest pip release:

[image]

Can you please check latest pip as well?

Please also note that there is safeguard feature. If there is ANY python exception in bCNC. Configuration is NOT saved upon shutdown to make sure it is not overwritten with errorneous config. So if you have any other issues causing bCNC exception unrelated to this, you will not be able to save config and you should restart bCNC and try to avoid that exception, modify config and restart bCNC again to make sure it is saved.

Harvie, 2021-02-25:
Please reopen if you find any issues with latest release.

closed 2021-02-25 by Harvie