vlachoudis/bCNC #1269: Probe Camera disable crosshair

Marthinius, 2019-08-28:
Hello,
Does anyone know how to turn off the camera crosshair ? 
And second problem...
I'm trying to create a custom button that hides the left control panel. Is it even possible ?

I've been looking for a few hours and I haven't found anything on this topic.

Thanks for the great program

Marthinius, 2019-09-09:
Nobody knows ? Is it possible ?

Harvie, 2019-12-01:
It should be possible to change its color. But i don't think it can be disabled.
However it seems to me like it would be easy feature for beginner to implement, if you are willing to contribute, i can merge your code. It's something like reading key from configuration file (which is already loaded to array in code) and then just put "if" conditional before the code that draws the crosshair.