vlachoudis/bCNC #1271: bCNC blocks when spindle motor turns on

Toscano2019, 2019-08-29:
Hello
my cnc works perfectly when the spindle motor is off. When I turn it on and start the execution of the g-code, bCNC stops immediately or at most after a few seconds.
This is what Python reports:

Controller state changed to: Idle (Running: False)
Exception in thread Thread-2:
Traceback (most recent call last):
  File "c:\Python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "c:\Python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "c:\Python27\lib\site-packages\bCNC\Sender.py", line 657, in serialIO
    self.mcontrol.viewStatusReport()
  File "c:\Python27\lib\site-packages\bCNC\controllers\_GenericController.py", line 87, in viewStatusReport
    self.master.serial.write(b"?")
  File "c:\Python27\lib\site-packages\serial\serialwin32.py", line 313, in write
    raise writeTimeoutError
SerialTimeoutException: Write timeout

Will clean after this operation

Does anyone know what happens?
Thanks

Harvie, 2019-08-29:
This sounds pretty much like EMI issue. What kind of spindle do you have? What kind of cables do you use? Send some photos of your machine :-)

closed 2019-08-29 by Harvie

MARIOBASZ, 2019-08-29:
Most likely, between electrical noise through the usb cable, it should be as short as possible, away from noise sources, and not parallel (not a bcnc problem)

Toscano2019, 2019-08-30:
Thanks for the answers.
some additional information:
Self-made CNC
shielded cable 4 wires awg26, shield not connected..
spindle motor https://it.aliexpress.com/item/32962395557.html?spm=a2g0o.placeorder.0.0.1b36580aOr8l52
I tried to separate the motor cable, which passed with the others, but does not change anything.
I enclose some photos of my masterpiece :-)

[image]
[image]
[image]
[image]

Be understanding .. it's the first one I make..

Toscano2019, 2019-08-30:
last thing..
Uno and CNC shield   https://it.aliexpress.com/item/32950052187.html?spm=a2g0s.9042311.0.0.27424c4dZPcZYP

Harvie, 2019-08-30:
Use shielded cable for spindle. I had this exact problem until i used shielded cable for spindle (shielded stepper cables didn't worked). Also switched to shorter USB cable and shorter cables in general. But shielded spindle cable did the trick.
Always connect the shields of shielded cables! There is very little that shielded cable can do if it's not properly grounded.

You need to setup your machine in a way that it has one central grounding point and all shields and grounds meet there:

[image]

[image]

[image]

[image]

It might be preferable not to connect shields on the other hand, so you never create ground loop, which might act as coil picking up EMI.

Google "cnc shielding" or "cnc star ground" and stuff like that, you will find lots of info.

MARIOBASZ, 2019-08-30:
At the end of the race, noise may also appear: try to disable hard limits, use soft limits enabled for safety, the end of the race is useful for homing.
Control the grounding issue

Toscano2019, 2019-08-30:
Thank you all. I set to work to put into practice what was recommended.
I will let you know the results

bobbynohairs, 2019-08-31:
Sounds like spindle motor EMF interference to me.

Toscano2019, 2019-09-04:
I used a shielded cable for spindle motor. Now works fine. thanks all

tavolito, 2019-11-10:
Hola Toscano tengo el mismo problema , me podrias mandar foros del tipo de cable que usastes , puede ser a mi mail o wattsapp 0054 1149796722     buengueroffgustavo@hotmail.com  desde ya muchas gracias

Harvie, 2019-11-10:
@tavolito I use **helukabel JZ-HF-CY 4G1** for spindle

tavolito, 2019-11-10:
Gracias en la semana comienzo a buscarlos