vlachoudis/bCNC #1272: Debugging--single block?

jrlandau, 2019-08-30:
Newbie question:  Is there a way to execute a g-code program one line at a time?  And when a program stops, how do you find out where the error occurred?  Is there a forum for questions like these?  Thanks.

MARIOBASZ, 2019-08-30:
One opinion: If it were possible to add line numbers with a button, it might be easier to track roughly until the line has been done.

Harvie, 2019-12-01:
It's not currently possible. But it would be great feature to have. We are all craving for something like this. So it's likely that we will see some features like this in near future. We just need to figure out how to implement this stuff, because inner workings of bCNC are currently not prepared for this.