vlachoudis/bCNC #1273: Tool Change suggestions.

DeanRM, 2019-09-04:
Why the config button?
My understanding for the button is to capture the difference between Z zero and the touchplate.  The only time this needs to be done is when workplace Z zero is reset.  This step is necessary and you are relying on the user to remember to do it.
Suggestion.
Remove the buttons, add a enable autocalibrate flag in the config file and auto run the calibrate macro every time Z is zeroed.  This macro would only run when needed at the start of sending the gcode.

Second suggestion - 
Add a Resume button on the tool change tab that is only enabled during a change.  This button would replace the need to go back to control tab and hit the Pause button.  (It does the same function.)  [You might want to add a Abort button also - the same function as Stop]

Harvie, 2019-12-01:
Sounds interresting. I currently don't have setup with tool length probing capabilities, so it's not likely that i will get to implement this anytime soon. But i will happily review and possibly merge such improvements if someone does this.

DeanRM, 2019-12-01:
How do I get started?