vlachoudis/bCNC #1274: Question: Detect error in macro, and display warning?

BernardG, 2019-09-06:
I have this macro working on a button for an XYZ probe. It works perfectly well, BUT, if I attempt to detect corners too close from the machine limits, the macro stop without any error, it just does not do the motion which would risk to go beyond the limits. Not complaining about this, I just would like to be able to trap the error and warn the user. I am sure GRBL send an error code, just don't remember which one.

If you need the macro's code, I will be glad to share it.

BernardG, 2019-09-11:
Surprised there are no answers, yet. Maybe I should phrase that differently: Is there a way, for a macro, to trap an error? Obviously, bCNC do it, as the "message bar" change color. So, how could I do, to warn users of the problem?

Harvie, 2019-12-01:
if you use the `spy://` entries in serial connection, the bCNC will log all serial communication to commandline (given that you've launched bCNC from commandline, not just clicking on the icon). This might help you to debug the problem.

[image]

BernardG, 2019-12-01:
Thanks for this, I did not know. But that does not really solve the problem. 
In my code, I have a dialog like:
```
%_mydia=app.entry("Entrez le diametre:");
%_demidia=float(_mydia)/2;
and so on ....
```
If the user don't enter a value, _mydia is empty and it gives me immediately an error, but the whole code continue to be executed. I am looking for a way to trap this error to exit without going through the rest of the code.

Harvie, 2019-12-02:
have you tried something like this?

%break
%exit
%exit(0)

BernardG, 2019-12-02:
No? I have not. I have to read about this to see what would be the best way to use your idea.

MARIOBASZ, 2019-12-03:
Maybe I write nonsense, or maybe it serves as an idea:
Do something similar to the plugins?
An alarm can be set to interrupt its operation:
if mydia == ""
return