vlachoudis/bCNC #1275: Question: If Python code was inserted in a G-Code file, would bCNC execute it, like in a macro?

BernardG, 2019-09-06:
I guess it's all in the subject. I bet the answer is no, but that would really be great, as it would add Macros possibilities to GRBL. Any chance it could happen?

Harvie, 2019-09-06:
Yes.

closed 2019-09-06 by Harvie

MARIOBASZ, 2019-09-07:
Sometimes I am very clumsy, I do not understand what you mean: Yes I know that you can generate a macro in Gcode, as is the case in canned operations for drilling, or in M6 to change tools (for now in development for personal use, I am adding the sentences for an automatic tool changer).
You mean this guy is macro, generating an M code, or can you clarify the idea?

BernardG, 2019-09-07:
What I was asking is, if I insert Python code in a Gcode file, like, say
%_myVar = 254;
then use _myVar later, would bCNC accept it and make it work, and Harvie answered yes! Which is GREAT!

GRBL does not allow Gcode macros per se, but this is a step toward it. Still no way to do a loop, as far as I know.

MARIOBASZ, 2019-09-07:
That's good, I'm curious to see the direction you take.
Repeating lines sounds interesting for example. G98 G99 yes you can from bcnc.

BernardG, 2019-09-08:
I am not sure yet how I will (or not!) use this possibility, but I am (for now, from time to time, but I will be more consistent this fall) working on implementing Fusion 360 Probe cycles into GRBL Post-Processor. What exists for now (in Fusion 360) has been written for Renishaw probes (and mostly Haas machines), so the code generated is of no use to help me here!

Maybe it's a bit ambitious, and I have a lot to learn, but I believe it would be great to make that work!