vlachoudis/bCNC #1277: Wiki Autolevel Notes

Guenni75, 2019-09-08:
Hello.
I don't understand a thing in the Autolevel Wiki.

Notes 4.:
The probe.......  data, no need to click "Autolevel". _**If you click the "Autolevel" button**_, the gcode is modified with the leveling data before sending to the controller, so if you have the measured data saved, the correction is then applied twice and you will ruin your board!

Which Autolevel-Button??? 
There is the Probe Tab, where i can switch to Autolevel. 
But this Button doesn't modify Gcode.
And the scan Button doesn't change anyhing too.
When does bCNC changes the GCode?
I never saw bCNC modifying my Gcode, especially not at AutoLevel.

Guenni75, 2019-09-08:
AHhh ok

closed 2019-09-08 by Guenni75

Harvie, 2019-09-24:
If you have ideas how to make information in wiki more clear for newbies, please don't hesitate and update it. You should have access to do so.