vlachoudis/bCNC #1281: Tool change (TLO) possible bug?

BernardG, 2019-09-13:
[image]
 I guess everything is on the picture.

Here is the process I followed before. I put a quite long tool, did my workpiece Z zero, then put the probe on the table, BELOW the workpiece Z zero, and calibrated. Calibration value is -13.944.

Did G54 to check, and the tool was just right at G54 X0 Y0 Z0.

Then I did change the tool to a shorter one. After the process completed, instead of going back to Z0, the tool went down to Z-20.664, as shown in the picture. Does not seems right.

Am I doing something wrong, or is this some bug?

Harvie, 2019-12-01:
The longer the tool, the larger your Tool Length offset will be.

There might be some problem related to these: #1140 #1035 #1208
We haven't yet figured out what to do in such cases, wheter reset the GRBL completely or not after the job. I am not really sure how this should be handled properly... Any ideas?

BernardG, 2019-12-01:
Let me get back to test this, I'll let you know my ideas. What I am not sure to understand is why this cause problem, while WCS works just fine. In both cases, TLO is calculated, no? It's just not used the same way. Or do I miss something?

Harvie, 2019-12-02:
These offsets and various coordinate systems were always kinda confusing to me, so i tend not to use them, but it would be cool to figure it out... Problem is that there are LOTs of their combinations and even more use cases and workflow. I always fix these for one guy and another pops out saying that i've broken it for him. So first we have to agree on how is this supposed to work.

BernardG, 2019-12-02:
I see what you mean! -:)
About how it "should'" work: The current WCS system is just fine as it is. You have to setup Z0 for WCS on the work piece, then, when you change tools it does check Z0 again. For me, it does really work as expected.

For TLO, you should have to setup ANOTHER Z0 than the work piece, and calibrate it. Then, it should work just as above, that is, when tool is changed, it's measured against the same reference as the "other" Z0 was. To me, it looks like TLO's problem is that it still use G54 Z0, while it should use G53 (or maybe G59). I don't know the internal code of bCNC for this, but I would not be too surprised if changing the coordinate system for TLO is all that is required to make it work (well, at least make it easier to get it to work)

In fact, the real goal (and use) for TLO is to have a fixed probe, somewhere on the machine, which would always be used for tools change, while WCS would be the proper method for those who do not have a secondary probe (secondary to finding the part Z0, or X0Y0Z0)

Harvie, 2019-12-03:
What do you think about changes in #1140 ?

DeanRM, 2020-03-20:
Here is my two cents worth - to install a fixed touchplate, you have to modify settings in the config file.  See the bottom of #1371 on how to do this.  This may or may not help with the TLO issue here.