vlachoudis/bCNC #1282: Manual Tool Change only work for the first occurence of M06 command

plapointe6, 2019-09-13:
I am using the Manual Tool Change (WCS) feature withing the last version of bCNC (master branch of this repository cloned a week ago), but it seems to work only the first time a M06 gcode command is encountered. For exemple, if there is multiple M06 instructions (ex: multiple hole sizes) in the gcode commands, the tool change will work the first time, but the others will be ignored and the gcode instructions will continue with de wrong tool.

Here is what I am doing : 
1- In my case, it's for PCB milling. I create one gcode file for each job with FlatCAM (one for isolation routing, one for each drill size and the last one for the board cutout).
2- I manually add a M06 command at the end of each gcode file except for the last one (board cutout).
3- Within bCNC, I open the first job to execute (ex: isolation routing).
4- Right after, I use File->Import to import all the other jobs, one by one.
5- I now have my complete PCB printing setup separated with M06 commands (I verified that all M06 commands are here when I save the merged gcode and I open it).
6- I start the milling process.

BernardG, 2019-09-13:
Hum, maybe that's a dumb question, but in your G-Code, are the different tools identified by different numbers? I ask, because when I started with Fusion 360, I did not pay attention to this, all my tools where N°1, and obviously, bCNC did not "see" any tool change, beside the 1st...

plapointe6, 2019-09-13:
Hum, pretty sure that they are identified with the same number ...
Dumb error, my bad !
Thanks for your help @BernardG 

I will retry with different numbers, but I am pretty sure that my issue is resolved.

closed 2019-09-13 by plapointe6