vlachoudis/bCNC #1284: File "Reloading" without opening

brettlx, 2019-09-24:
Hi. Thanks for a great application. Currently running on Pi.
The only issue I have is with the loading of some files. 
Granted these are large in size (over 4MB) as they are engravings so not sure if that is the issue.
But when I load the file via the application it says reloaded (even new files which I have never opened before) and does nothing. However, if I upload them via the Pendant webpage they load correctly.

Screenshot of having uploaded via Pendant:
[image]

Screenshot of "Opening" the same file via a local folder on the Pi:
[image]

As can be seen, there is nothing in the editor to indicate that anything loaded.
I did restart bCNC between loads to ensure there was nothing currently loaded.

The other obvious difference is that the one is local and the other is being uploaded from my PC. Not sure if this would make a difference. Could this be a local file permissions issue or something like that? I don't have the same issue with all files. Some small nc files loaded every time from the local path.

brettlx, 2019-09-24:
Okay, please ignore. Looks like it was just a file permissions issue. 
However, could you please explain the "reloaded" message when nothing actually happens (file read error I'm guessing).

Harvie, 2019-09-24:
bCNC automaticaly reloads g-code file if you modify it. Eg. if you change some parameters in external CAM software and save the g-code under the same name, bCNC will automaticaly reload it for you.