vlachoudis/bCNC #1286: Z-values not correct in G-code

ITDSIvansMill, 2019-09-25:
Dear all,
now that I am retired, I finally have some time to pick up CNC milling. I installed bCNC ( version 0.9.14 - 5feb2018) under Windows 10 and it seems to me a fine working program with a lot of possibilities. However, after creating material ( with feed= 70, plunge feed=30, depth increment= 0.2), stock (safe Z=1, surface Z=0), and loading a dxf file from Inkscape, the machine just hoovers over the material and does not mill. I went through almost all 1000 open and closed issues here on GitHub but can not find a similar problem. That's why I am asking if someone please can help me.
When I look at the G-code in the editor, it looks for all blocks something like this 

g0 x15.129226 y286.41869 z1
g0 x15.129226 y286.41869 z0
(entered)
g1 z0 f30
g1 x40.153412 y286.37903 z0 f70
g1 x15.129226 y286.41869 z0
(exiting)
g0 z1

So, the machines travels fast with safe Z=1 to coordinates, lowers to the surface (Z=0) and states "entered". To me this is not correct since entering is in the next command "g1 z0 f30". But this entering does not change the Z value ( I expected here z-0.2=depth increment instead of z0). This explains why the machine just hoovers over the surface. Of course, I could trick the system by setting the surface Z value to the milling depth, but this can not be the solution (also it then breaks my bits of 0.5 mm diam since it is driving too fast into the material). I should also mention that I tried the "cut plug-in" but this gives the same behaviour of not entering the material correctly.

Can please someone tell me what I am doing wrong ? I don't think its the program since nobody mentiones this problem in the issues, so it has to be me. But after several days of trying a lot of things, I am now out of ideas.

MARIOBASZ, 2019-09-25:
the code is milling at z = 0
If Z target = -2.3 and step 0.2
select block
form:
type cut -2.3 0.2 below in command editor
safe speeds and height will be according to selected material and stock
If I remember correctly, if you only put cut.
Then you can experiment with conando tab, cut command, profile, etc.

ITDSIvansMill, 2019-09-25:
Dear Mariobasz,
thanks very much for your quick reply. 
I followed your suggestions, It takes indeed values from material and stock, but the problem is that the surface Z is ignored. Here are some of the first lines of the code

g0 x15.129226 y286.41869 z1
g0 x15.129226 y286.41869 z-0.2
(entered)
g1 z-0.2 f30
g1 x40.153412 y286.37903 z-0.2 f70
g1 x40.153412 y261.473822 z-0.2
g1 x15.049911 y261.473822 z-0.2
g1 x15.129226 y286.41869 z-0.2
g1 z-0.4 f30
g1 x40.153412 y286.37903 z-0.4 f70

As you can see, the second line is wrong: it should refer to z0 instead of z-0.2.  All the rest looks fine. By this way the machine will be lowering the head at high speed into the material and so breaking the thin bit.

Harvie, 2019-09-25:
Ignore "stock" and "material" for now. Just use "cut" and show us screenshot of all your settings of that module. Also you might want to use "profile" or "pocket" before "cut" to compensate for cutter radius. Also make sure you didn't enabled "laser mode" anywhere. I think that disables Z movement, but i am not sure...

ITDSIvansMill, 2019-09-25:
Dear Harvie, I will try to upload some screenshots. This is the first time in my live I am doing this, so hope all works well.
This is the setting of the cut parameters :
[image]

And this is what the editor window shows of the code. I unfolded the first block where you can see that the second line should refer to z0 instead of z-0.2.
[image]

In the config window I see that the laser mode should be turned off:
[image]

I hope this is sufficient info for you.

I am also starting to think that it is a bug in the program. Normally you do not detect this problem unless you are working with very thin mills and the milling head raised high above the material.

Harvie, 2019-09-25:
You have old bCNC version. Please install latest using **pip** as described in README and wiki.

MARIOBASZ, 2019-09-26:
test with 0.2 deep increment (positive)

ITDSIvansMill, 2019-09-26:
Sorry for the late reply, i had yesterday evening somenone come over to install the latest version with PIP since i don't trust myself in doing this ( seems to be now version 0.9.14-dev).
This morning i went through the new features and get some python errors in the terminal window ( like "Bitmap bCNC.ino not defined" when i push about button)  and an error when pushing the error report button.

But, i find these errors not so important since i tried also some milling and my PROBLEM IS SOLVED !!
The cutting CAM works just great !

**Thanks to all of you for the fast replies and usefull entries !**

I can now start exploring this version and start milling with thin diameters without always breaking the bits.
For me this topic can be closed.

Harvie, 2019-09-26:
> like "Bitmap bCNC.ino not defined" when i push about button

Never seen this. Maybe some error with bCNC.ini file? You can try deleting it (will loose all your setings of bCNC and it will get automaticaly generated with clean settings. this will not affect GRBL settings)

> seems to be now version 0.9.14-dev

Yes. actually it is development version, but we are moving fast to stable release, so right now i would consider this version the best.

> i had yesterday evening somenone come over to install the latest version with PIP since i don't trust myself in doing this

Did they showed you how to do it? Once the pip is set up correctly (which is, because you managed to install bCNC), it should be easy to upgrade bCNC every time you need.

Just do `pip2 install --upgrade bCNC` if there is new version available and it will be downloaded and installed (which is currently not, since you've already installed it yesterday)

closed 2019-09-26 by Harvie