vlachoudis/bCNC #1287: Linearize Plugin Failure - ZeroDivisionError

rkburnside, 2019-09-28:
I am trying to use the "linearize" plugin and am getting the error below. This is happening on:

bCNC-0.9.14.30-beta-git-exe.zip
bCNC-0.9.14.52-beta-git-exe.zip

I have also downloaded and run the source code and still get the same error. I have also reinstalled python several times and that hasn't fixed the issue.

Any idea what could be causing it?

```
new-config Utils <ConfigParser.ConfigParser instance at 0x034ABC60>
Traceback (most recent call last):
  File "bCNC\Utils.py", line 460, in __call__
  File "bCNC\ToolsPage.py", line 1740, in execute
  File "C:\Users\richa\Downloads\CNC etch a sketch\bCNC\plugins\linearize.py", line 69, in execute
    npath = opath.linearize(maxseg, splitlines)
  File "bCNC\lib\bpath.py", line 802, in linearize
  File "bCNC\lib\bpath.py", line 280, in linearize
ZeroDivisionError: float division by zero
```

Harvie, 2019-09-28:
Please use pip version, exe release is not stable yet.

rkburnside, 2019-09-29:
OK...I installed bCNC using PIP and had the same issue. It did provide a bit more information (see below). 

`Traceback (most recent call last):
  File "C:\Users\richa\Anaconda2\lib\site-packages\bCNC\Utils.py", line 464, in __call__
    return self.func(*args)
  File "C:\Users\richa\Anaconda2\lib\site-packages\bCNC\ToolsPage.py", line 1747, in execute
    self.tools.getActive().execute(self.app)
  File "C:\Users\richa\Anaconda2\lib\site-packages\bCNC\plugins\linearize.py", line 69, in execute
    npath = opath.linearize(maxseg, splitlines)
  File "C:\Users\richa\Anaconda2\lib\site-packages\bCNC\lib\bpath.py", line 803, in linearize
    linearized.extend(seg.linearize(maxseg, splitlines))
  File "C:\Users\richa\Anaconda2\lib\site-packages\bCNC\lib\bpath.py", line 285, in linearize
    linearized.append(Segment(Segment.LINE, self.distPoint(i*step), self.distPoint((i+1)*step)))
  File "C:\Users\richa\Anaconda2\lib\site-packages\bCNC\lib\bpath.py", line 239, in distPoint
    return self.extrapolatePoint(dist)
  File "C:\Users\richa\Anaconda2\lib\site-packages\bCNC\lib\bpath.py", line 226, in extrapolatePoint
    raddist = dist/self.radius
ZeroDivisionError: float division by zero
`

Harvie, 2019-09-29:
Ok, can you please send me the file that you are trying to linearize and the settings of linearize tool that you are using.

rkburnside, 2019-09-30:
settings:
Name: (left blank)
segment size: 1.0
subdiv lines: (unchecked)

[circle-arcs.txt](https://github.com/vlachoudis/bCNC/files/3668425/circle-arcs.txt)

Harvie, 2019-09-30:
Fixed in bCNC-0.9.14.62
You can do `pip2 install --upgrade bCNC`

closed 2019-09-30 by Harvie

rkburnside, 2019-10-01:
just tried it out and the fix works! thanks...I appreciate your help.

commit 0582f774 (2019-09-30): Linearize arcs with zero radius. Fixed #1287