vlachoudis/bCNC #1289: Scripting: How to wait last command to finish to get final XYZ position?

rasciodc, 2019-10-05:
Not really an issue, but a question:

Using the scripting on the user macro buttons I'm running this code:
`G91 G38.2 X-5 F100
%xl=float(wx)
G91 G38.2 X5 F100
%xr=float(wx)
%center_x=(xl+xr)/2
G91 G0 X(center_x)
G92 X[0]
`

Harvie, 2019-10-05:
have you tried `%wait` ?

rasciodc, 2019-10-05:
> 
> 
> have you tried `%wait` ?
> https://github.com/vlachoudis/bCNC/wiki/Scripting

Yes, I did, just like this:

    G91 G38.2 X-5 F100
    %wait
    %xleft=float(wx)
    G91 G38.2 X5 F100
    %wait
    %xright=float(wx)
    %center_x=0.5*(xleft+xright)
    G91 G0 X(center_x)
    G92 X[0]
 It did not work... =/

Harvie, 2019-12-01:
There seems to be some issue with %wait, see #1304 we probably need to fix this

rasciodc, 2020-02-28:
> 
> 
> There seems to be some issue with %wait, see #1304 we probably need to fix this

What is the status on this? Is there any update? (I'm genuinely curious, not trying to be pushy or demanding anything lol) 

I saw at least other 2 users trying to do exactly what I was intending and don't know if they succeeded. Cheers!

rgodin974, 2020-06-19:
Hello, did you find a solution with %wait ? I would want like to do this kind of macro too. Thanks

rasciodc, 2021-05-30:
> 
> 
> Hello, did you find a solution with %wait ? I would want like to do this kind of macro too. Thanks

I still haven't any solutions to that... Has anyone?