vlachoudis/bCNC #1290: Plugin Question: Is it possible to un-check a checkbox from the execute() method?

mj1911, 2019-10-07:
Hello, I'm trying to modify the sample plugin. I would like to let the user check certain boxes for the tasks they wish performed.  When the plugin is executed, the checkboxes should be cleared. Is there a way to programatically clear these checkboxes from within the def execute(self, app)? I have tried:

```
self["CheckBox"] = 0 # no error, but does not work.
self["CheckBox"]["bool"] = False # TypeError: 'int' object does not support item assignment
self.variables["CheckBox"] = 0 # TypeError: list indices must be integers, not str
self.variables["%s.%d"%("CheckBox",self.current)] = 0 # TypeError: %d format: a number is required, not NoneType
self["CheckBox"].set(False) # AttributeError: 'int' object has no attribute 'set'
self["CheckBox"].bool = False # AttributeError: 'int' object has no attribute 'bool'
self.variables.CheckBox = False # AttributeError: 'list' object has no attribute 'CheckBox'
self.variables.CheckBox.set(False) # AttributeError: 'list' object has no attribute 'CheckBox'
self.variables["CheckBox"] = False # TypeError: list indices must be integers, not str
self.variables.index["CheckBox"] = 0 # TypeError: 'builtin_function_or_method' object does not support item assignment
self.variables.index["CheckBox"].set(False) # TypeError: 'builtin_function_or_method' object has no attribute '__getitem__'
self.variables.index["CheckBox"].index["bool"] = False # TypeError: 'builtin_function_or_method' obj has no attrib '__getitem__'
self.variables[5] = False # no error, but no work. gave an error when re-checking the box, however.
self.variables[5][:2] = False # TypeError: 'tuple' object does not support item assignment
self.variables[5][2] = 0 # same
self.variables[5:2] = 0 # TypeError: can only assign an iterable

global config	# from Utils.py
config.set(variables, "CheckBox", str(int(0))) # NameError: global name 'config' is not defined
setbool(variables, "CheckBox", str(int(0))) # NameError: global name 'config' is not defined
```

Hopefully this functionality is present and I'm just clueless as to the exact wording.  Thank you.

mj1911, 2019-10-17:
Ok so after getting debugging working in the whole module, I learned that:

```
self["CheckBox"] = True   # changes value, but not checkbox state
self["CheckBox"] = False  # changes value, but not checkbox state
```
Is there a way to propagate this value back to the control, so the checkbox actually changes?  There are so many tkinter entries at run-time I'm having difficulty even finding the control.

Harvie, 2019-12-01:
Can you please describe exactly what the usecase would be? Sounds bit funny to me.

mj1911, 2019-12-03:
Sure. Plugin can generate a circle, a rectangle, and a triangle. User checks "circle", and enters size of 5mm, coordinates, etc.  Then the execute button is pressed and gcode is created.  User then checks "rectangle", changes size to 20mm, and clicks execute.  However they forgot to uncheck "circle", so now TWO operations will be performed (20mm circle and 20mm rectangle.)  These (circle, rectangle, triangle) checkboxes must be automatically un-checked after every execute to prevent this.

Harvie, 2019-12-03:
This sounds like use case for selectbox, same as i use in cutting strategy:

[image]

mj1911, 2019-12-17:
That is a very good solution. Unsure why I was so fixated on using checkboxes for this.  Thank you for the suggestion.

Harvie, 2019-12-17:
No problem. Thanks for your interrest in bCNC development :-)

closed 2019-12-17 by Harvie