vlachoudis/bCNC #1292: drawing bot DXF import Z safe / retract settings

GradeSchool, 2019-10-09:
I'm importing .dxf files to use for a home-made drawing bot. Everything works, but my problem is that I need to set the Z safe / retract height manually (or override the default values- which in my case is set to 3 mm when I examine the gcode). 

The retract value of 3 mm is just not enough to clear "bumpy" paper. I need to set it higher- like around 8 mm.

Where are the default values set? I've scoured the UI and the github, but can't find what I am looking for. I don't want to manually edit the gcode if I don't need to. Thanks!

Harvie, 2019-10-09:
Look to the "CAM->stock" menu.

closed 2019-10-09 by Harvie

GradeSchool, 2019-10-09:
Gotcha! Thanks very much for the quick response.

GradeSchool, 2019-10-09:
Sorry Harvie- I changed the setting in CAM- stock from 3 to 10mm, then restarted bCNC. My setting in bCNC is correct, but after importing the gcode- I examine the gcode andthe settings are still set to Z safe of 3 mm. Any thoughts?

GradeSchool, 2019-10-09:
OKay it finally worked- just took a few retries! Close bCNC, restart, etc.

Harvie, 2019-12-01:
bCNC will not save the bCNC.ini if there was some error message printed to console during the operation, so it does not corrupt the data. But if you launch the bCNC, edit the parameter and close bCNC without doing anything else which could have caused python error, it should save.