vlachoudis/bCNC #1293: Split and Join Not Working / Giving Errors

rkburnside, 2019-10-10:
I get the following error when I try using split / join:

Traceback (most recent call last):
  File "C:\Users\richa\Anaconda2\lib\site-packages\bCNC\Utils.py", line 464, in __call__
    return self.func(*args)
  File "C:\Users\richa\Anaconda2\lib\site-packages\bCNC\CNCList.py", line 689, in splitBlocks
    bl = Block(self.gcode[bid].name())
  File "C:\Users\richa\Anaconda2\lib\site-packages\bCNC\CNC.py", line 2171, in __getitem__
    def __getitem__(self, item):                return self.blocks[item]
IndexError: list index out of range

My cnc file is attached. BTW, I am using the latest version installed through PIP. Thanks for the help!
[goat.txt](https://github.com/vlachoudis/bCNC/files/3710173/goat.txt)

MARIOBASZ, 2019-10-10:
On my machine, joining works correctly with your example.

Harvie, 2019-10-10:
Works for me too. However i use Linux.
Can you please do `pip install --upgrade bCNC` two or three times? I've noticed sometimes the old version gets cached and does not install at the first try... Current pip version should be 0.9.14.62.

However even at older versions i never had this problem. Maybe something is different on windows... What version of windows do you use?

rkburnside, 2019-10-11:
- ran "pip install --upgrade --force-reinstall bCNC" and got the same result.
- installed it on another computer and got the same result.
- installed and ran using a fresh install of python and got the same result
- installed anaconda2 and got the same result.
- I'm running windows 10 on all my machines.

MARIOBASZ, 2019-10-11:
I use Windows

MARIOBASZ, 2019-10-11:
I use Windows 7

Harvie, 2020-04-17:
@rkburnside Can you please share g-code which causes this (possibly with screenshots) ? Perhaps this is some edge case...

Harvie, 2021-02-23:
Closing this, i will reopen if you will provide the g-code for testing.

closed 2021-02-23 by Harvie