vlachoudis/bCNC #1295: how work bCNC ?

manojprasanga, 2019-10-18:
How work bCNC source code cnccanvas code?

Harvie, 2019-10-18:
What exactly do you mean?

manojprasanga, 2019-10-18:
I need to understand bCNC cnccanvas source code

manojprasanga, 2019-10-18:
Is any explanation about source code of bCNC in the internet or other things ?

Harvie, 2019-10-18:
> I need to understand bCNC cnccanvas source code

Why? Are you trying to improve the bCNC? If you tell me exactly what are you working on, we will be able to help you.

manojprasanga, 2019-10-19:
Sir, Actually i am working on cnc simulation project .So, i need to develop Canvas like in bCNC canvas .So, try to take a idea about bCNC canvas

Harvie, 2019-10-19:
OK. I would strongly reccomend you to NOT use CNC canvas from bCNC. It's VERY SLOW and i am very sad about it. For large g-code files it can take hundreds of seconds to refresh screen. It's the WORST part of bCNC and i am looking to replace it with some faster OpenGL based screen.

If you are looking into CNC simulation, take a look here:

https://camotics.org/
http://www.cncwebtools.com/Apps/GCode_Simulator/index.html
https://www.freecadweb.org/wiki/Path_Simulator

closed 2019-10-19 by Harvie

manojprasanga, 2019-10-19:
Sir, i need to create my own simulator.

Harvie, 2019-10-19:
I know, but don't use CNC Canvas from bCNC for this purpose. It will make your simulator slow and bad. Trust me, it's not a good idea.

manojprasanga, 2019-10-19:
Thank you sir

Harvie, 2019-10-19:
Maybe if you are looking for some 3D canvas in pyhton, take a look at something like this:

https://docs.enthought.com/mayavi/mayavi/

There are like 10 similar 3D canvases for use with python.