vlachoudis/bCNC #1297: AttributeError: 'NoneType' object has no attribute 'split'

montagsmodell, 2019-10-27:
I try to setup bCNC on Ubuntu 18.04.; With the command $ python2 -m bCNC in folder ~/Documents/bCNC-master/build/lib$ I get the following error:

`
montagsmodell@Kepler:~/Documents/bCNC-master/build/lib$ python2 -m bCNC
new-config Utils <ConfigParser.ConfigParser instance at 0x7fc6b601e248>
Using fallback Utils.comports()!
self.canvas.winfo_id(): 62914656
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/montagsmodell/Documents/bCNC-master/build/lib/bCNC/__main__.py", line 2708, in <module>
    main()
  File "/home/montagsmodell/Documents/bCNC-master/build/lib/bCNC/__main__.py", line 2676, in main
    application = Application(tk)
  File "/home/montagsmodell/Documents/bCNC-master/build/lib/bCNC/__main__.py", line 196, in __init__
    page = cls(self.ribbon, self)
  File "/home/montagsmodell/Documents/bCNC-master/build/lib/bCNC/CNCRibbon.py", line 115, in __init__
    self.register()
  File "/home/montagsmodell/Documents/bCNC-master/build/lib/bCNC/FilePage.py", line 392, in register
    (SerialFrame,))
  File "/home/montagsmodell/Documents/bCNC-master/build/lib/bCNC/CNCRibbon.py", line 134, in _register
    w = f(self.master._pageFrame, self.app)
  File "/home/montagsmodell/Documents/bCNC-master/build/lib/bCNC/FilePage.py", line 253, in __init__
    self.comportRefresh()
  File "/home/montagsmodell/Documents/bCNC-master/build/lib/bCNC/FilePage.py", line 347, in comportRefresh
    for hw in i[2].split(' '):
AttributeError: 'NoneType' object has no attribute 'split'

`
Does anyone has the same issue and can help me set it up please?

TGit-Tech, 2019-10-29:
I'm getting the exact same error.
Windows 10 - Just followed exactly the Windows install procedure on the wiki.
Python version 2.7.17

A fix or workaround would be great;  I'm still searching.

Harvie, 2019-12-01:
Yes, the serial/pyserial situation is bit of a mess. Since the pyserial is drop in replacement for serial, but i think bCNC needs both installed in right order, or something like that. I am aware of this situation... I think i will add some tests to check the pyserial installation and inform user if there's something wrong with it...

treki1, 2020-07-09:
After upgrade to LinuxMint 20 mate I get the error " AttributeError: 'NoneType' object has no attribute 'split' "
(Before that I had to install numpy.)
Any suggestions?