vlachoudis/bCNC #1298: Won't detect camera with Linux / Raspberry

andrecalais, 2019-10-30:
Dear Vasilis,

How are you?

First, thank you for your contribution with the bCNC software, it's a superb tool and very professional.

I'm currently facing a problem with Linux while running it. The software starts fine and the front end is working well, but it can't detect the camera.

By "not detecting the camera", I mean all the buttons that toggle camera on/off are disabled, so is the camera configuration button in the CAM section.

I ran a code in python to detect my camera and display it and it worked. So, there's no problem with python and it's packages.

I executed with terminal and no errors were displayed. On Windows and Mac the software runs perfectly, detecting the camera and all. Unfortunately I have to make it work with Linux... 

My bCNC version is 0.9.14.

Have you ever faced any problems with Linux? Heard or helped someonelse with the same problem?

Thanks in advance!

Att., Andre Calais Salles.

Harvie, 2019-12-01:
Have you tried to go to the camera settings and change camera number? Maybe your system detects multiple video inputs and bCNC is choosing the first one by default, which is not the right one.

andrecalais, 2019-12-01:
> Have you tried to go to the camera settings and change camera number? Maybe your system detects multiple video inputs and bCNC is choosing the first one by default, which is not the right one.

Yes, I tried all numbers detected by the system.

Thanks for answering!

Harvie, 2019-12-01:
Do you have python opencv module installed?
spoiler alert: not sure if it's available on ARM processors.

andrecalais, 2019-12-01:
Yes. I had all modules installed. I was able to run a python code with opencv to display the connected camera and it worked.

andrecalais, 2019-12-01:
> Do you have python opencv module installed?
> spoiler alert: not sure if it's available on ARM processors.

Yes. I had all modules installed. I was able to run a python code with opencv to display the connected camera and it worked.

Harvie, 2019-12-01:
Honestly i don't know what is going on. I am not author of the camera code and i don't have time to debug this. But if you figure it out, i will be happy to review and merge the fix :-)

That might be easy, since you already got it working in your python script, so just check what bCNC does differently. Sorry for inconvenience.

andrecalais, 2019-12-01:
No problem, I appreciate the effort. Thanks!