vlachoudis/bCNC #1299: Autolevel problem (maybe my mistake)

AuBeTeX, 2019-10-30:
HI!

I have a chinese 3108 CNC milling machine. i try to milling PCB. I read lot of web pages, wikis and try to not confusing. I'm failed. 

What i doing wrong? 

My steps:

1, Loading the file.
2, set the origin
3, Probe on origin position set z to 0
4, read margins from file with margins button
5, start scan.
6, Scan done, go back to origin
7, Probe on origin postition set z to 0 
8, Use the zero button, with z in origin postition and value is 0 the tool 
    touch on surface
9, Z move up 1 mm
10, Start milling.

What i doing wrong?
 I present a few pictures about my problem.
[image]
[image]
[image]

yrabbit, 2019-10-30:
On Wed, Oct 30, 2019 at 04:32:37AM -0700, AuBeTeX wrote:
> HI!
> 
> I have a chinese 3108 CNC milling machine. i try to milling PCB. I read lot of
> web pages, wikis and try to not confusing. I'm failed.
> 
> What i doing wrong?
> 
> My steps:
> 

The circuit board is pretty much curved. I would increase the number 
of steps along X and Y to about 30.

-- 
  with best regards,    
  Yellow Rabbit @yrabbit@mastodon.sdf.org
  DragonFly 5.5-DEVELOPMENT x86_64

robughblah, 2019-11-01:
Even if your bed is level, your bit may not be aligned to it. The bed, and the pcb will have a warp or twist in it.
You need to do an autolevel when your pcb blank is in place which overcomes these issues. 

And start out with a very simple circuit design till you get these issues resolved. They go faster. 

Here are some youtubes of autoleveling with bcnc.
https://youtu.be/Z59BkvBOx5A

https://youtu.be/O-PMukVrfOo

https://youtu.be/PG4IbFTXYeY

Harvie, 2019-12-01:
I usualy prepare g-code for pcb etching using tiny sharp v-bit so it cuts in zero depth. And i will then autolevel it and use contact probing using endmill to set Z zero. When i am in that Z zero contact point, i usualy set it to be something like 0.01mm, which then causes the endmill to cut 0.01mm in the copper. If that's not enough, i go 0.01mm deeper and re-run the g-code. 0.1mm is too deep and will result in teared copper traces, so always go in 0.01mm increments!

I've found this workflow to work the best for me...

Harvie, 2019-12-01:
Also read the wiki https://github.com/vlachoudis/bCNC/wiki/AutoLevel
and make sure you are zeroing the autolevel data to be relative to your touch probing point. Using special zero button in probe menu.

closed 2019-12-01 by Harvie