vlachoudis/bCNC #1301: Issue with driller pluggin

dodolbysuround, 2019-11-03:
Hello @HomineLudens, Sorry for bothering but I think you are the guy for this question:

I'm using the Driller pluggin for laser burn 1 dot at every single end of straights segments imported as a DXF with "use anchor" option checked (I want no point along any segment, only at both ends)
I face 2 littles problems generating the Gcode;
1-) 1st dot of every serie is burnt twice
2-) the last dot of the last serie is not burnt

Any idea where it could come from?

here is an exemple with 3 series of segments.
[image]

the parameters I use
[image]

here are 3 points marked twice
[image]

here is the last point that is missing
[image]

MARIOBASZ, 2019-11-04:
Can you copy the file you want to run driller on?

dodolbysuround, 2019-11-04:
here you are

[test segment simple.zip](https://github.com/vlachoudis/bCNC/files/3802947/test.segment.simple.zip)

HomineLudens, 2019-11-04:
Hi @dodolbysuround I'll try to give a look. When I wrote the plugin laser mode was not supported yet. Maybe it is not using the right functionality.

MARIOBASZ, 2019-11-04:
@HomineLudens , Since I read you, I thank you for your contributions: together with the anchor addition they were the starting point to generate my complements.
@dodolbysuround , Can you confirm if you first do a first drill at x = 0 y = 0?
I have changed the following in the driller plug-in and corrected both problems and it worked correctly:
online 232:
# if idx> 0:
   if idx> 1:
online 234:
# newHolePoint = (anchor [0] [0], anchor [0] [1], anchor [0] [2])
   newHolePoint = (anchor [1] [0], anchor [1] [1], anchor [1] [2])

MARIOBASZ, 2019-11-04:
(choose end of vector instead of start, skip first move)

MARIOBASZ, 2019-11-04:
I made a mistake, in milling mode the last drill misses the reference height

dodolbysuround, 2019-11-04:
I get this error when opening bCNC when I try your modifications
[image]

MARIOBASZ, 2019-11-04:
I made a mistake, I'll try again

MARIOBASZ, 2019-11-04:
On my pc Driller with anchor works well in an old version of bcnc.
In recent versions, if it is the first block in the list or below the header, it generates the first drill at the point x = 0 and = 0, it generates twice at the first point (it does not make the exception for movements only on the Z axis) , however there is no difference in the plugin.

MARIOBASZ, 2019-11-04:
Any changes in b path or other file that generates the problem?
I have made a change in my anchor machine: ignore if the movement is vertical.
It would work by adding an exception if the selected block is number 0 or if it is the first block after header, but I don't think it's a neat solution

dodolbysuround, 2019-11-05:
> (choose end of vector instead of start, skip first move)

> Hi @dodolbysuround I'll try to give a look. When I wrote the plugin laser mode was not supported yet. Maybe it is not using the right functionality.

hooo yes pleaaaaase...

dodolbysuround, 2019-11-13:
Hi @HomineLudens  ,
I don't want to be ennoying but do you think you'll find some time to look at the driller pluggin in laser mode?
I'll have some heavy drawings to handle and I'm scared I'll need to correct them "by hand"... sniff...

HomineLudens, 2019-11-14:
Hi @dodolbysuround I'm very sorry but I have my hands full and I don't think I could work on this any time soon.
Hope someone can help you.

Harvie, 2019-12-01:
I have some plans to do major rewrite of driller code. Currently there are two or three driller tools in bCNC (including the pre drill X marking using laser), so i want them to be merged into single one. But if someone is willing to do this cleanup for us, i will be happy.

dodolbysuround, 2019-12-26:
> I have some plans to do major rewrite of driller code. Currently there are two or three driller tools in bCNC (including the pre drill X marking using laser), so i want them to be merged into single one. But if someone is willing to do this cleanup for us, i will be happy.

Hi @Harvie ,
Thank you so much for your efforts.
I just tried the plugin, and it turns out that:
- M3 / M4 selection works well
[image]

- drills generation works well if "M3 for laser" is checked BUT
- there are still troubles when I try it with  "M3 for laser" unchecked: The very first point to be drilles is shouldn't be drilles, the first point of each segment is drilled twice, and the very last point that should be drilles is not. 
[image]

Can you give it a check?

Thank you by arvance

GitHubCaps, 2020-01-26:
@dodolbysuround , I have fixed the issue, but not ready to release it yet. (see below)

`test segment simple M3.nc`
M5
F3000
G0 M3 S12000
G0 X40 Y-10.0001 F3000
G1 X40 Y-10.0001
G4 P0.1
G0 X10.0001 Y-40 F3000
G1 X10.0001 Y-40
G4 P0.1
G0 X35.0001 Y-35.0001 F3000
G1 X35.0001 Y-35.0001
G4 P0.1
G0 X29.9999 Y-40 F3000
G1 X29.9999 Y-40
G4 P0.1
G0 X35.0001 Y-25 F3000
G1 X35.0001 Y-25
G4 P0.1
G0 X25 Y-35.0001 F3000
G1 X25 Y-35.0001
G4 P0.1
G0 X20 Y-20 F3000
G1 X20 Y-20
G4 P0.1
G0 X29.9999 Y-10.0001 F3000
G1 X29.9999 Y-10.0001
G4 P0.1
G0 X14.9999 Y-14.9999 F3000
G1 X14.9999 Y-14.9999
G4 P0.1
M5

I have a few questions concerning:
> Which would be the easiest way to get "M3"s instead of "M4"s when using Driller plugin in laser mode?
This is because I would prefer constant laser power while GRBL $=32 laser mode unableles (in order to keep being able to use dynamic mode for other operations)

Do you mean that you don't want to change modes manually and just want to choose the option for M3 as I provided in the plugin? Grbl does not know which laser mode `M3` or `M4` is used.

Just for clarification,

`Laser mode enable`, in `Controller`:
Switch to turn on laser cutter (disables spindle)
Controller must be updated for the machine to work in laser mode.

[image]

`Laser Cutter`, in `Config`:
Enables `M3` inside of `bCNC` only...

`Laser Adaptive Power`, in `Config`:
Enables `M4` inside of `bCNC` only...

For encoding `M3`/`M4`  into the `gcode` file, nothing to do with `grbl` controller.

[image]

With that being said, are you using `Laser Adaptive Power` `M4` for normal use? If not, then there is no reason for  `Laser Adaptive Power` to be ticked in `bCNC`/`CAM`/`Config`. I don't know your settings, just trying to figure out the best way to help you. Would you like to also draw the line in `M4`, while marking the point in `M3`, since `M4` won't mark it without a motion?

Again, for clarity, are you attempting to use `M4` for marking with the laser? `M4` will not work for this unless a motion, such as drawing a circle, is programmed in the plugin. Since the laser will only fire during motion, `M3` is the only mode you can use for a point unless you want a ring mark. I hope that makes sense, just figured I would point that out in case you didn't know that.

In short, the controller doesn't care about `bCNC`'s config settings. You can actually use `M3`/`M4` together in the same `gcode` file, so long as the plugin or whatever you are using to code it with is setup properly, hence why I would like to know what you are trying to accomplish with `M4` or if you didn't realize that it will not work to begin with without other modifications. Either way, once I am finished with adding all drill plugins together, it shouldn't matter, because it should be able to handle both scenarios with a few clicks.

@Harvie ,  I am in the middle of adding all the drill plugins (`drill` from `ToolsPage.py`, `driller.py`, `drillmark.py` and also @DodoLaSaumure `SimpleDrill.py`) together as one, probably within the next few days. Hopefully, you can shed some light on the subject.

1. What are your thoughts about adding the option for appending `grapid` feeds, is it even necessary (for `grbl`), or just keep for compatibility reasons?

2. What about `cutfeed`/`cutfeedz`? I believe that this is useful, rather than having to switch back and forth between `material`

3. `Peck`? If `None` (left blank) or `0`, turns it off, but maybe a boolean option to use `material` `Depth Increment`

I don't want to put in to many options that aren't necessary, making it too confusing for others. At the moment, anything left blank uses `stock`/`material` database as defaults.

Also, note that I added an option (see image) to `Start at id number`, which may help to resolve issues (such as in this post) about starting at `g0 x0 y0`. It may alleviate the problem, just by increment `> 0`, better than it being hard coded into the file.

Adding  `Drill ignore G1 moves` as an option, also seems more fitting instead of hard coded in the file.

Added `Driller fix anchor point` @MARIOBASZ as option, instead of changing it's normal function.

> Any changes in b path or other file that generates the problem?
> I have made a change in my anchor machine: ignore if the movement is vertical.
> if idx > 0:
> if (anchor[0][0] != anchor[1][0] and anchor[0][1] != anchor[1][1]):
> newHolePoint = (anchor[0][0],anchor[0][1],anchor[0][2])
> bidHoles.append(newHolePoint)
> newHolePoint = (anchor[1][0],anchor[1][1],anchor[0][2])
> bidHoles.append(newHolePoint)
> 
> It would work by adding an exception if the selected block is number 0 or if it is the first block after header, but I don't think it's a neat solution.
> This problem could affect another complement.

Work in progress:
[image]

Best regards, GitHubCaps

DodoLaSaumure, 2020-01-26:
> What are your thoughts about adding the option for appending grapid feeds, is it even necessary (for grbl), or just keep for compatibility reasons?

For me giving this option offers more flexibility to the end user

> What about cutfeed/cutfeedz? I believe that this is useful, rather than having to switch back and forth between material

I totally agree with you

> Peck? If None (left blank) or 0, turns it off, but maybe a boolean option to use material Depth Increment

I would not set a boolean, but a value in mm of the peck seems to be more flexible to me

A great thank for your works on this project
Have a good day
Pierre

GitHubCaps, 2020-01-26:
> I would not set a boolean, but a value in mm of the peck seems to be more flexible to me

The boolean if used, just automatically uses `Depth Increment`. So it would just be an option for those who don't want to fill in the values and instead rely on there `stock`/`material` settings. Same goes for leaving `cutfeed` and `rapidfeed` blank. Notice all the blank entries in the image, all defaults were pulled in from `bCNC`'s configuration.

[image]

Your welcome. Hopefully, the next release solves all the issues.

DodoLaSaumure, 2020-01-26:
@GitHubCaps  OK Great
Thank you 
Have a good day

MARIOBASZ, 2020-01-26:
I did not read in detail, but in proncipio I think the following:
1- laser mode would believe that enabling / disabling from grbl is sufficient (never use laser, I may be wrong) and is responsible for turning it off for G0
2- The line appears on my screen
G0 M3 S1200:
should only appear
M3 S12000
online next
G0 x40 and-10,0001 F3000
should be separated into two lines to
G0 x40 y-10.001
F3000
(excuse me ... F3000 is not very fast? at least for bits)
In non laser mode
I would add a quick drop G0 to improve times significantly hadta 1mm before contact with the surface and then G1 (maybe you've already done it)
3- I am not sure if my modification in anchor solved the problem

MARIOBASZ, 2020-01-26:
I don't understand checkboxes in
fast feed add,
nor what does it mean
fast feed xy
fast feed z.
Opinion consultation: for multiple holes larger in diameter than the milling cutter, should it be placed as an option there, and call a helical, or is it preferable to do this in the helical complement only?

GitHubCaps, 2020-01-26:
@MARIOBASZ 

> 1- laser mode would believe that enabling / disabling from grbl is sufficient (never use laser, I may be wrong) and is responsible for turning it off for G0

Correct. Disabling from the controller is all that is needed.

> G0 M3 S1200:
should only appear
M3 S12000

Either way works. `G0` is a safety measure for using the laser, basically keeps it from firing for compatibility with older software. No harm done.

> G0 x40 and-10,0001 F3000
should be separated into two lines to
G0 x40 y-10.001
F3000

Wrong. We are appending rapid feed rate. We are dealing with laser in `M3` mode. A single Fxxx at the top of the file sets the rate for cutfeed.

> (excuse me ... F3000 is not very fast? at least for bits)

You can set that to F5000, but it will still only go as fast as the setting in the controller, so if you have it set as 2000 in the controller, then that's all the feedrate you are going to get.

> In non laser mode
I would add a quick drop G0 to improve times significantly hadta 1mm before contact with the surface and then G1 (maybe you've already done it)

Not sure what contact you talking about with the laser, only using X and Y axis at the moment, hence the way the file in the post is setup.

> 3- I am not sure if my modification in anchor solved the problem

Yes. It does help to have it as an option, but I also left the original way intact.

> I don't understand checkboxes in
fast feed add,
nor what does it mean
fast feed xy
fast feed z.

It is optional. That is the the appended rapidfeed, not something you normally get, but many people want or request it.

F3000                                (cutfeed)
G0 M3 S12000                  (set the power for the laser, this is not the spindle rate for milling)
G0 X40 Y-10.0001 F3000 (rapidfeed to this point)
G1 X40 Y-10.0001            (know we burn the mark)
G4 P0.1                             (for 0.1 seconds)

An example of how I accomplish this
		# Use plugin settings, otherwise fallback to stock/material database
		try:
			self.cutfeed = int(self["cutfeed"])     # moveFeed
		except:
			self.cutfeed = int(CNC.vars["cutfeed"])
		try:
			self.cutfeedz = int(self["cutfeedz"])   # drillFeed
		except:
			self.cutfeedz = int(CNC.vars["cutfeedz"])

		self.appendRapid = self["appendRapid"] # bool
		try:
			self.feedmax_xy = int(self["feedmax_xy"])
		except:
 			# use the min value for safety, but it still depends on settings in the controller
			self.feedmax_xy = int(min(CNC.feedmax_x, CNC.feedmax_y))
		try:
			self.feedmax_z = int(self["feedmax_z"])
		except:
			self.feedmax_z = int(CNC.feedmax_z)

> Opinion consultation: for multiple holes larger in diameter than the milling cutter, should it be placed as an option there, and call a helical, or is it preferable to do this in the helical complement only?
Maybe mark values, type of operation in helical (holes, threading) and option in drill boolean box use helical?

Not sure I know what you mean there. Helical? In drill mode, we are using peck or going to the depth in one shot. The drill, driller, drillmark plugins do not support helical. They were/are meant to act more like a boring machine with drill bits.

MARIOBASZ, 2020-01-26:
Ok

GitHubCaps, 2020-01-26:
@MARIOBASZ , see the couple of edits above, please.

dodolbysuround, 2020-01-26:
Wohhhhh what an impreprissive amount of interesting posts!

@GitHubCaps Thank you for clarifications about:
> Laser mode enable, in Controller:
> Laser Cutter, in Config:
> Laser Adaptive Power, in Config:

I was aware about the 2 first but I do recognize I missed the last one...  maybe this might be documented somewhere (either on the wiki or in contextual message boxes. 

> With that being said, [...] just trying to figure out the best way to help you. 

This is very kind from you and I really apreciate

> Would you like to also draw the line in M4, while marking the point in M3, since M4 won't mark it without a motion?

You couldn't guess better.  That's exactly what I mean to do. However, the way I do it changed a little bit since the begining of my tests:

- At the begining, to adjust the hole diameter, I used to increase/reduce the G4 PX parameter in M3 mode. But it turned out the thicker the paper is, the less efficent is the method to increase hole diameter. Then I found out the work around proposed by @MARIOBASZ :

> +1: Opinion consultation: for multiple holes larger in diameter than the milling cutter, should it be placed as an option there, and call a helical

I personally accomplished it with an Excel macro by modifying the bCNC generated .nc file because I didn't want to bother you again... but from my opinion, that would be an interesting feature! In my case, I used circular motion instead of helical one because my holes are very small and I work with laser cutted paper si I don't have to care about X/Y pass increment.

[image]

[image]

For each hole, my gcode is as follow:
`g0 x9.7014 y31.287 F3000 (rapid to specific position)
G1 S12000 (fire the laser)
g91 (change to relative moves)
g03 x0.20 i0.10 (first circular movement to go from the center of the circle to the desired radius)
g03 x-0.40 i-0.20 (first half circle)
g03 x0.40 i0.20 (second half circle)
g90 (back to absolute moves)
G1 S0 (turn the laser off` 

- For drawing the lines, There is no doubt M4 is much more efficient that M3. 

> Again, for clarity, are you attempting to use M4 for marking with the laser? M4 will not work for this unless a motion, such as drawing a circle, is programmed in the plugin. Since the laser will [...]

Once again thank you for this huge work. You do not imagine how helpfull and convenient bCNC is for me thse days since I discovered these driller pluggins.

One last question: Is the bugg @MARIOBASZ  speaks about a bit earlier about vertical and horizontal moves has been fixed?

Good luck for puting all of this all toogether!

GitHubCaps, 2020-01-27:
> Wohhhhh what an impreprissive amount of interesting posts!

@dodolbysuround , sometimes, that's what it takes to get to the heart of the matter!

> Thank you for clarifications

Your welcome. I figured it wouldn't hurt to mention a few things that may not be obvious at first glance, lol.

> I was aware about the 2 first but I do recognize I missed the last one... maybe this might be documented somewhere (either on the wiki or in contextual message boxes.

I don't think it's documented well in `bCNC`, however `grbl` has some decent info, though still a bit vague.

[image]
[https://github.com/gnea/grbl/wiki](url)

[image]

The above command can be done in `bCNC` at the command line without going to the `controller`.
In `bCNC` `config`, the laser options are a convenience function for scripting, ie. it automatically uses `M3`/`M4` to make it easier for the users. It is very handy, but it does not solve everything for you. It still depends on what program or plugin is being used and its intended purpose. For example, I believe there's only two plugins that are even intended to be used with a laser. `Drillmark` (for manually marking) and the other is `Halftone` (for working with raster images). Aside from that, the actual laser setting in the `config` basically just supplies an `M3` for `milling` (if `false`) with appropriate `G0`/`G1` values where necessary, or `M3` for `lasercutter` (if `true` and not `laseradaptive`), `M4` for `laseradaptive` (if `lasercutter` is `true`) and doesn't actually have anything to do with `grbl`. Only thing that `grbl` knows about is whether or not it's in `milling` (normal) mode or `laser` mode. 

[image]

Notice the `TODO`. Yes, someone _needs_ to put up decent suggestions!

[image]
[https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode](url)

This link has some decent info: [https://wiki.shapeoko.com/index.php/G-Code](url)

One thing to note is the lower value for decimals. 3 decimals is plenty for accuracy in `mm` and easily changed in `config`.

> This is very kind from you and I really apreciate

I appreciate that you appreciate!

> You couldn't guess better. That's exactly what I mean to do. However, the way I do it changed a little bit since the begining of my tests:

Haha, my tests have also been changing a lot. For some reason, I figured it was a bit more complex than just some simple lines with holes needing punched! If you can give me better details, such as your workflow, format (cad/cam, dxf, nc, raster images, etc) you are starting with and if you could upload a better file to produce the expected outcome would be nice.

> At the begining, to adjust the hole diameter, I used to increase/reduce the G4 PX parameter in M3 mode. But it turned out the thicker the paper is, the less efficent is the method to increase hole diameter. Then I found out the work around proposed by MARIOBASZ
+1: Opinion consultation: for multiple holes larger in diameter than the milling cutter, should it be placed as an option there, and call a helical

What is the purpose of the hole, or better yet, what size are you trying to create? If I understand you correctly, you need to pierce through the paper and that is why you are increasing diameter. I am not sure, either way.

> I personally accomplished it with an Excel macro by modifying the bCNC generated .nc file because I didn't want to bother you again...

If you can upload the file for testing, i may be able to get better results.

Don't worry about bothering me, because I was going to apologize about not getting back to you sooner. I was gone for a few weeks and didn't know that the problem persisted until I got back. Part of my plan this year is to give back to the bCNC community for such an awesome program! Without the devs and the users input, we wouldn't have what we have today.

What did you use to produce the following code?
```g0 x9.7014 y31.287 F3000 (rapid to specific position)
G1 S12000 (fire the laser)
g91 (change to relative moves)
g03 x0.20 i0.10 (first circular movement to go from the center of the circle to the desired radius)
g03 x-0.40 i-0.20 (first half circle)
g03 x0.40 i0.20 (second half circle)
g90 (back to absolute moves)
G1 S0 (turn the laser off
```
> For drawing the lines, There is no doubt M4 is much more efficient that M3.

`M4` is definitely a nice feature. They both have their strengths and weaknesses, just depends on the task and materials at hand.

> One last question: Is the bugg @MARIOBASZ speaks about a bit earlier about vertical and horizontal moves has been fixed?

I have added his previous fix as an option, plus I added another option so that the user can manual adjust the start point in the file, so hopefully between the two fixes there won't be anymore problems.

After all of this, I have figured out that the best solution is probably not to rely on` bCNC`'s config for what mode it is in and just give the user the option: 

Milling            (normal mode)
LaserM3        (M3 only)
LaserM3+M4 (M3 for burn point, M4 for drawing line)

In the end, I believe that's the way to go.

Thanks, now back to coding!

dodolbysuround, 2020-01-28:
Okay, hence, lets try to explain my project...
the ultimate goal is to produce paterns for embroidery works.
that consists in drawing a wireframe picture on a kind of Canson paper (the lines drawn in M4 mode) and make very tiny holes (0.35mm of diameter is my target) to show where the needle.
As I explaines previously, my initial plan was to adjust the hole diameter wy the g4 pX parameter in steady x/y position in M3 mode. this method works for certain type of paper but for some reasons, doesn't for other types (with high quality paper, iven if you increase "exporure" time the hole doesn't get biger) that's hy I changed strategy and staster to burn holes by the mean of describing a smell circle around the theorical position of the hole in M4 mode. 

now, the workflow: 
starting from a DXF file imported in bCNC:

1. I use the `CAM/Driller` to pre generate the Gcode for the holes
2. I save it in .NC
3. I use my home made excel file to over the .NC
4. I re-open the overwritten .NC file in bCNC
5. I use the `CAM/Cut` tool to generate the gcode for the lines

I gess you'll laugh when readind the macro because this is probably not rigousous way to code still, that does the job...

Hope my explainations are lear...

do not hezitate to tell me if you do not seiz anything.

[attachement.zip](https://github.com/vlachoudis/bCNC/files/4124687/attachement.zip)

MARIOBASZ, 2020-01-29:
I appreciate each contribution, I continue learning from each one.
Are you looking to generate holes / circles of a specific diameter (for example 0.35mm) over a route, separated by a distance X?
If someone doesn't do it before, I will try to insert anchor and helical laser function.

dodolbysuround, 2020-01-29:
Are you looking to generate holes / circles of a specific diameter (for example 0.35mm) over a route, separated by a distance X?
@MARIOBASZ, I am not very sure to understand you r question but the distance between 2 holes is never the same.
Holes has to be placed at each end of every single segment ( the inpit DXF is only composed of segments)

GitHubCaps, 2020-01-29:
@dodolbysuround Thanks, for the attachment.  Now, I can really help you out. I dont't think you need the absolute to relative mode. However, that would probably make it easier for you to copy/paste blocks (holes) wherever you need them, correct?

GitHubCaps, 2020-01-29:
I will try and set it up so that you can re-run the file (if necessary) with just repeating the holes (no need to pass over the lines again.) I believe that that would probably be more suitable to your situation. Possibly, even better to have it create two files and auto-save. One complete with lines and holes, the other with just holes.

dodolbysuround, 2020-01-29:
> I don't think you need the absolute to relative mode
indeed, it is pretty convenient because I may have to slightly modify the position of one hole or add another one manyaly. then I just need need to add a `Go Xxx.xxx Yyy.yyy` and copy past the "replacement code" delow.

> I will try and set it up so that you can re-run the file (if necessary) with just repeating the holes (no need to pass over the lines again.) I believe that that would probably be more suitable to [...]
I don't see what you want to say...

> Edit: I guess, you could just run the dxf in M4 mode, but I still like all-in-one solutions.
Indeed, now, I only use M4 **for this application** but I think M3 G4 Ps is interesting in oteher cases.

Another feature that could be added in my opinion is search/replace. For exemple, if I find out I set the wrong feed rate (Ffff) or laser poser (Ssss), so far, I 

1. save in .NC, 
2. open it with block note, 
3. search and replace what I need to replace, 
4. and re open in in bCNC. 

Not very straight foreward isn't it?

This is a bit far away from the initial topic but I think this could be an interesting feature for bCNC... What do you think?

GitHubCaps, 2020-01-29:
>  I may have to slightly modify the position of one hole or add another one manyaly. then I just need need to add a Go Xxx.xxx Yyy.yyy and copy past the "replacement code" delow.

From a quick look at your file, the line segments appear to be about 5mm. How far apart should the holes be? One at each end and one in the center (2.5mm)?

> One complete with lines and holes, the other with just holes.
I don't see what you want to say...

What I am saying is that I can make it produce two files on-the-fly. One complete with everything, the other with just holes in case you need to run it again (the file with only holes). That can save you time from editing.

> I only use M4 for this application but I think M3 G4 Ps is interesting in oteher cases.

That will be an option, to also include the `G4` during the same run as `M4`. Simple supply a dwell time greater than 0 and it will automatically turn on `M3` to burn at the setting. Best of both worlds scenario. `M4` will do the lines and circles and `M3` will stop and burn for the set time.

> Another feature that could be added in my opinion is search/replace. For exemple, if I find out I set the wrong feed rate (Ffff) or laser poser (Ssss)

Haha, maybe. However, all you will need to do is select a block for testing with your initial settings on scrap material. Run the file and adjust settings until you get the results you want, then select all your blocks (main `dxf`, of course) to do the real thing.

I will set it up for different feedrates and spindle power for lines versus holes. Even the power for `M3` (burn just the mark) will be independent of the other settings. That will help you out a lot, also if you do need to do a search and replace, they will be of different values (depending on how you set), example: `F1000` `S12000` for `M4` line,  `F900` `S11000` for `M4` hole, `F800` `S10000` for `M3` burn mark with `G4 P0.1`

This is a bit far away from the initial topic but I think this could be an interesting feature for bCNC... What do you think?

I think as I pointed out above, with the new settings I will have in the file, simple re-run the original `dxf` again with different settings and you will be all done. This will hopefully be an all in one solution to keep you from having to do manual edits at all.