vlachoudis/bCNC #1302: flatten.py->class Flatten->#raster approch->while loop->#up double appending of points

simcup, 2019-11-04:
not sure if this is a bug but it looks like you're appending the points of the flaten pattern twice after the first run. are the xP.append and yP.append after the z step there purposefull?

simcup, 2019-11-04:
just checked by commenting out the appending of points after counting up acctualY, gcode has no more double lines

Harvie, 2019-12-01:
Is this about the flatten plugin? I mean this one?

[image]

Would you mind making a pull request if you found out how to fix the issue?

GitHubCaps, 2019-12-22:
Instead of commenting out, just add a check for last position to keep from duplicating the line.
Commenting out will indeed cause adverse effects instead, at least during my quick test.