vlachoudis/bCNC #1303: New Interface - maybe Off Topics

onekk, 2019-11-09:
Only for knowledge.

Some time ago there was a discussion on interface changes for small screens, especially for thos using RPi and similar computers to run bCNC.

There was also a discussion on integrating a "Industrial control like"  on the jog panel.

After some work, I've managed to made some changes.

But the codebase is derived from bCNC but with some mayor changes, deleting some not useful code and modifying many things.

So I have forked bCNC some days ago (about a week) and intergate my changes on the codebase, I have removed all the reference to bCNC (hoping having done it right) in the code and progrma. leaving an appropriate credit also on my github page, having not much fantasy I have renamed it OKKCNC.

I wish to thanks all the developers of bCNC 

Best Regards to all.
Carlo D.

Harvie, 2019-11-09:
I am excited.
Can you please share the link?
Did you managed to fix the performance issues with graphical rendering of big files?
Can you share the details about deleting some not useful code and modifying many things?

onekk, 2019-11-09:
The code is in my personal repo, follow my name and you will find it, maybe here is not the "correct" place to discuss.

Regards

Carlo D.

closed 2019-11-09 by Harvie