vlachoudis/bCNC #1307: Problem with very very small values

MARIOBASZ, 2019-11-20:
If the value is close to zero, it may be that the result obtained leads us to a wrong location, for example:
x 7.9335 e-06, the machine will go to x-7.9335 instead of zero.
It would be good a general solution to serve for all cases and plugins

MARIOBASZ, 2019-11-20:
in this case the value obtained was from the center plugin

Harvie, 2019-12-01:
Maybe we need to round() the value in center plugin...

MARIOBASZ, 2019-12-01:
Rounding does not solve the problem.
It is a problem of number format, avoid scientific notation.
As a solution instead of
str ("g1x") + str (x)
I have used
str ("g1x") + format (x, 'f')
however, I think cnc.gline already deals with the problem

Harvie, 2019-12-02:
I've fixed the center plugin, is everything OK for you now?

commit 494ec2b2 (2019-12-02): Fixed #1307
closed 2019-12-02 by Harvie