vlachoudis/bCNC #1310: Spanish problem init bCNC

cavaret, 2019-11-27:
Python 2.7.15+

./bCNC
###########################################
WARNING! THIS IS LEGACY MODE!
DO NOT USE THIS IF YOU ARE A BCNC DEVELOPER
GO TO REPOSITORY ROOT AND LAUNCH BCNC USING
FOLLOWING COMMAND:

python2 -m bCNC
###########################################

new-config Utils <ConfigParser.ConfigParser instance at 0x7f42c061b9e0>
self.canvas.winfo_id(): 73400416
Using old style comports()!
Using old style comports()!
Using old style comports()!
/home/josemiel/bCNC/bCNC/ProbePage.py:1876: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  CNC.toolWaitAfterProbe = int(TOOL_WAIT.index(self.toolWait.get()))
Traceback (most recent call last):
  File "./__main__.py", line 2713, in <module>
    main()
  File "./__main__.py", line 2681, in main
    application = Application(tk)
  File "./__main__.py", line 196, in __init__
    page = cls(self.ribbon, self)
  File "/home/josemiel/bCNC/bCNC/CNCRibbon.py", line 115, in __init__
    self.register()
  File "/home/josemiel/bCNC/bCNC/ProbePage.py", line 1988, in register
    (ProbeCommonFrame, ProbeFrame, AutolevelFrame, CameraFrame, ToolFrame))
  File "/home/josemiel/bCNC/bCNC/CNCRibbon.py", line 134, in _register
    w = f(self.master._pageFrame, self.app)
  File "/home/josemiel/bCNC/bCNC/ProbePage.py", line 1667, in __init__
    self.toolWait.set(TOOL_WAIT[1])
  File "/home/josemiel/bCNC/bCNC/lib/tkExtra.py", line 3094, in set
    self.invoke()
  File "/home/josemiel/bCNC/bCNC/lib/tkExtra.py", line 3046, in invoke
    self.command()
  File "/home/josemiel/bCNC/bCNC/ProbePage.py", line 1876, in waitChange
    CNC.toolWaitAfterProbe = int(TOOL_WAIT.index(self.toolWait.get()))
ValueError: u'ANTES y DESPU\xc9S del sondeo' is not in list

Harvie, 2019-11-28:
Can you please try latest git master. Also run it properly using `python2 -m bCNC`. Not using legacy script.

cavaret, 2019-11-28:
the problem persists, the problem is in the tildes since they removed the .encode ('utf8')
#-----------------------------------------------------------------------
	def waitChange(self):
		CNC.toolWaitAfterProbe = int(TOOL_WAIT.index(self.toolWait.get().encode("utf8")))
		CNC.toolWaitAfterProbe = int(TOOL_WAIT.index(self.toolWait.get()))

python -m bCNC 
new-config Utils <ConfigParser.ConfigParser instance at 0x7f7c0f2d2950>
self.canvas.winfo_id(): 39845984
Using old style comports()!
Using old style comports()!
Using old style comports()!
ProbePage.py:1876: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  CNC.toolWaitAfterProbe = int(TOOL_WAIT.index(self.toolWait.get()))
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/josemiel/bCNC/bCNC/bCNC.py", line 2713, in <module>
    main()
  File "/home/josemiel/bCNC/bCNC/bCNC.py", line 2681, in main
    application = Application(tk)
  File "/home/josemiel/bCNC/bCNC/bCNC.py", line 196, in __init__
    page = cls(self.ribbon, self)
  File "CNCRibbon.py", line 115, in __init__
    self.register()
  File "ProbePage.py", line 1988, in register
    (ProbeCommonFrame, ProbeFrame, AutolevelFrame, CameraFrame, ToolFrame))
  File "CNCRibbon.py", line 134, in _register
    w = f(self.master._pageFrame, self.app)
  File "ProbePage.py", line 1667, in __init__
    self.toolWait.set(TOOL_WAIT[1])
  File "/home/josemiel/bCNC/bCNC/lib/tkExtra.py", line 3094, in set
    self.invoke()
  File "/home/josemiel/bCNC/bCNC/lib/tkExtra.py", line 3046, in invoke
    self.command()
  File "ProbePage.py", line 1876, in waitChange
    CNC.toolWaitAfterProbe = int(TOOL_WAIT.index(self.toolWait.get()))
ValueError: u'ANTES y DESPU\xc9S del sondeo' is not in list

Harvie, 2019-11-28:
BTW can you please try this in python3 as well? Just to compare...
Does the problem happen when you switch to english language or it's just in spanish translation?

cavaret, 2019-11-28:
el problema solo ocurre por la acentuación en el español en ingles no sucede

Harvie, 2019-11-28:
This is strange `CNC.toolWaitAfterProbe = int(TOOL_WAIT.index(self.toolWait.get()))`
It's a number entry. Why do you have spanish letters in the number entry box?

cavaret, 2019-11-28:
I just tried it in python3 and it doesn't happen

Harvie, 2019-11-28:
Good to know. Python 3 has overall better handling of unicode (national alphabets). I think bCNC development will gradualy move towards python3 as python2 gets deprecated at the end of the year by it's creators. But i will try to figure this out so it can work on both python2 and python3...

cavaret, 2019-11-28:
ok, thanks

gnurc, 2020-01-31:
Spanish language not works, the application crash when start. UTF problem.

Harvie, 2020-12-28:
Reportedly fixed in this PR. Please test and close this issue if everything is OK:
https://github.com/vlachoudis/bCNC/pull/1405#issuecomment-751580531

closed 2020-12-28 by cavaret