vlachoudis/bCNC #1311: Zooming is something wrong and result in erratic behaviour

onekk, 2019-11-27:
Hi, I0ve noted that using the fit2screen function sometimes, it behaves erratically, or better the scale is not respected, due to the fact that float rounding in the code:

```

        if zx > 1.0:
            self.__tzoom = min(zx, zy)
        else:
            self.__tzoom = max(zx, zy)

```

make the scale to became 0.99988 something and so it is < 0 and the max value is used

modifying thecode to assume some prudent controls correct the wrong behaviour

```

        # add a factor to improve reability
        bbox_width = (x2-x1) * 1.05
        bbox_height = (y2-y1) * 1.05

        try:
            zx = round(float(self.winfo_width() / bbox_width), 2)
        except:
            return

        try:
            zy = round(float(self.winfo_height() / bbox_height), 2)
        except:
            return

        print("BBCALC ", bbox_width, bbox_height)
        print("canvas ", self.winfo_width(), self.winfo_height())
        print("ZX, ZY ", zx, zy)

        if zx > 0.98:
            self.__tzoom = min(zx, zy)
        else:
            self.__tzoom = max(zx, zy)

```

Harvie, 2019-12-03:
How do i reproduce the problem? I hit F many times and nothing bad happens...

onekk, 2019-12-03:
The problem is float "internal representation", if you hit many times the fit 2 screen and the ratio is near 1.0 sometimes it could became 0.9999988 that did not match the > 1.0.

It is a strange, but expected behaviour, present, as seen in other issue of bCNC, it is related to the infamous IEEE floating representation, maybe it depends also on python versions, as some little change are done in internal math, as example in some 2.7 something the internal integer and long integer type were unified.

Don't worry, I noted the problem printing the zoom value after having noted the strange behaviour, if you select some paths and do zoom to fit the value is near 1.0 but there are some flashing in the resizing due to approximation errors, (print statements are clear).

For the tabs and space, as PEP8 spaces are the preferred way to indent python sources,  and many other bug are "present" in bCNC sources like the space around assignments and operator and after the commas (some source file raise 1400 PEP8 errors).

Regards

Carlo D.

Harvie, 2019-12-03:
I've tried again in python2 and it does lowkey weird stuff. if i press F multiple times it "vibrates" one pixel back and forth. It does not happen in python3.

Harvie, 2019-12-03:
Now i tried with your code and it happens as well. Same problem...

closed 2019-12-04 by Harvie

Harvie, 2019-12-04:
So i've merged this, because i actualy found cases, where your code worked better. Still needs some work tho. Eg.: sometimes i have to press F three times to get drawing fit properly...

onekk, 2019-12-05:
Yes the problem of the "vibrating" is generated by the calculation, i suppose the culprit is the bounding box calculation that generate float result but then this values are translated in pixels (integers), the zooming factor is not the same calculated in floating, maybe a different rounding in discrete steps, suppose to jump from 1.0 to 1.05 will gave some room to avoid boundaries values.

I have tryed to add a slightly border adding a factor 0f 1.05 to the bbox dimensions to obtain a better visualization (some room at border is better than the selected objects too near the borders).

suppose you obtain an actual zoom zx is 1.25 and after the new run the calculation is 1.26 or 1.24 the flickering is done instead if zx is kept 1.25 if it is >= 1.23 and <= 1.27 the 1.25 is retained.

I will search if there is a rounding routing in python, or a additional parameters to the round() function to obtain such discrete aprossimation.  Let me see.

Regards

Carlo D.

Harvie, 2019-12-05:
I don't think the virating is the biggest issue. When i open file it gets centered, then i hit center two more times and it jumps like 10+ pixels until it finaly gets centered correctly.

onekk, 2019-12-05:
Some preliminar analysis give me an hint about something is not working as expected in the **_zoomCanvas** function, I noted adding some line that the zooming factor seem to be not respected, the the **self.__tzoom** var you set to make the zoom is not a zoom factor, but a multiplier of the actual zoom factor **self.zoom**, but in the **fit2screen** function the actual zoom factor seems to be not taken in account, it is calculated as a proportion between the width and height of the **canvas** windows and the width and height of the bounding box of the selected objects, (but the real dimension are expressed in mm not in pixel and the mm to pixel conversion)  

this is the function in CNCCanvas that calculate bbox
```
    def selBbox(self):
        """Return selected objects bounding box"""

        x1 = None
        for tag in ("sel", "sel2", "sel3", "sel4"):
            bb = self.bbox(tag)
            if bb is None:
                continue
            elif x1 is None:
                x1, y1, x2, y2 = bb
            else:
                x1 = min(x1, bb[0])
                y1 = min(y1, bb[1])
                x2 = max(x2, bb[2])
                y2 = max(y2, bb[3])

        if x1 is None:
            return self.bbox('all')
        return x1, y1, x2, y2
```

it uses the Tkinter (the code is from Tkinter for python 2.7.13)

```
def bbox(self, *args):
        """Return a tuple of X1,Y1,X2,Y2 coordinates for a rectangle
        which encloses all items with tags specified as arguments."""
        return self._getints(
            self.tk.call((self._w, 'bbox') + args)) or None
```

if x1 is not specified it rely on calculating the bbox for **'all'** tag of the canvas.

but the representation you select is converted from mm to pixel (they are not real pixel but the canvas "unit of measure") by some other routines, they are "discretized", but to adapt them to the new zoom they are simply scaled using this code

```
        for i in self.find_all():
            self.scale(i, 0, 0, zoom, zoom)
```

I suspect that applying such scaling generate some misalignment that the subsequent recalculation triggered by another fit2screen press is modifying in sensible manner the proportion and a double click is needed to obtain a more correct aspect ratio, during the second run where the differences are less important.

But my knowledge of the internals of bCNC is not so deep to pinpoint all the aspect of the problem.

Regards

Carlo D.

Harvie, 2019-12-05:
I think this fit2screen code should be simplified as much as possible. Right now it feels unnecessarily complex to me... The basic idea is quite simple. isn't it? Is there really reason for this to be so complicated?

onekk, 2019-12-05:
I agree that the complications are many, but we are dealing with a complex piece of code.

found another bug in **zoomCanvas** no upper nor lower limit, if you insist with soom in and zoom out weird results happens, some correction are needed, maybe:

```
    def _zoomCanvas(self, event=None):
        """Zoom on screen position x,y"""

        x = self._tx
        y = self._ty
        zoom = self.__tzoom

        if OCV.GRAP_DEBUG is True:
            print("--- _zoomCanvas ---")
            print("ACTUAL_ZOOM", self.zoom)

        self.__tzoom = 1.0

        c_zoom = self.zoom

        c_zoom *= zoom

        if c_zoom > 30000:
            c_zoom = 30000
        elif c_zoom < 0.1:
            c_zoom = 0.1

        self.zoom = Utils.q_round(c_zoom, 4, 0.005)

```

with Utils.q_round

```

def q_round(x, prec=2, base=.05):
    """round a number specifing the decimal digits
    and a quantization factor"""
    return round(base * round(float(x)/base), prec)

```

About a simplified version of fit to screen, i think that retain the zoom factor and take it in account maybe is a solution, let me see some more.

Regards

Carlo D.

onekk, 2019-12-05:
A side note the upper and lower limits in zoom are needed to avoid some division by zero errors (for the lower) and for the upper a overflow leading somethig apper as string (i suspect a NaN or similar) that stops the mouse position to be calculated and raise error at every mouse movements.

commit 1cab1974 (2019-12-04): Better fit to screen code by Carlo "onekk" D.  Fixes #1311