vlachoudis/bCNC #1313: Potential problem in CNC.py

onekk, 2019-12-05:
This function in CNC.py class CNC 

```
    @staticmethod
    def gcode(g, pairs):
        s = "G{0}".format(g)
        for c, v in pairs:
            s += " {0}{1:0.{2}f}".format(c, round(v, OCV.digits), OCV.digits)
        return s
```
has the same name of **self.gcode**

```
        self.retractz = True  # G98/G99     retract to Z or R
        self.gcode = None
        self.plane = XY

```

I have renamed function **gcode** to **gcode_string** it is more descriptive on what it does and avoid ambiguity and potential errors.

Harvie, 2020-04-09:
Sounds reasonable... we can merge this if you are willing to make a PR

GitHubCaps, 2020-05-16:
@Harvie  Not reasonable, unless you want to recode all the plugins that rely on `CNC.gcode()`
@onekk  This actually is not even an issue, it is a `staticmethod` that is used by plugins for convenience.

You can place a `print` statement in there if you desire, it is honestly never touched by anything other than plugins. `self.gcode` is not the same thing in this instance.

GitHubCaps, 2020-05-16:
These are my own notes with an explanation of how I would like to improve on the code.
```python
	#----------------------------------------------------------------------
	# TODO: This function can be removed in favor of other function calls.
	# However, many plugins rely on it and they need to be recoded first.
	# Also, glinev and garcv use it; they should also be removed.
	# Note that we basically have three dupe funcs that do the same thing,
	# I would prefer the funcs that are more explicit and don't rely on
	# the zip function.
	# Used by plugins: e.g. CNC.gcode(1, [("f",CNC.vars["cutfeed"])])
	@staticmethod
	def gcode(g, pairs): # XXX: Issue #1313
		s = "g%d"%(g)
		for c,v in pairs:
			s += " %c%g"%(c, round(v,CNC.digits))
		return s

	#----------------------------------------------------------------------
	@staticmethod
	def _gcode(g, **args):
		s = "g%d"%(g)
		for n,v in args.items():
			s += ' ' + CNC.fmt(n,v)
		return s

	#----------------------------------------------------------------------
	@staticmethod
	def _goto(g, x=None, y=None, z=None, **args):
		s = "g%d"%(g)
		if x is not None: s += ' '+CNC.fmt('x',x)
		if y is not None: s += ' '+CNC.fmt('y',y)
		if z is not None: s += ' '+CNC.fmt('z',z)
		for n,v in args.items():
			s += ' ' + CNC.fmt(n,v)
		return s

	#----------------------------------------------------------------------
	@staticmethod
	def grapid(x=None, y=None, z=None, **args):
		return CNC._goto(0,x,y,z,**args)

	#----------------------------------------------------------------------
	@staticmethod
	def gline(x=None, y=None, z=None, **args):
		return CNC._goto(1,x,y,z,**args)

	#----------------------------------------------------------------------
	# TODO: This function should be removed in favor of 'gline' above.
	# Only box, midi2cnc and pyrograph plugins use it atm,
	# but they need to be recoded first.
	@staticmethod
	def glinev(g, v, feed=None):
		pairs = zip("xyz",v) if OLD else list(zip("xyz",v)) # XXX: Issue #1363
		if feed is not None:
			pairs.append(("f",feed))
		return CNC.gcode(g, pairs)

	#----------------------------------------------------------------------
	# TODO: This function should be removed in favor of 'garc' below.
	# Only the box plugin uses it atm, but needs to be recoded first.
	@staticmethod
	def garcv(g, v, ijk):
		if OLD:
			return CNC.gcode(g, zip("xyz",v) + zip("ij",ijk[:2])) # XXX: Issue #1363
		else:
			return CNC.gcode(g, list(zip("xyz",v)) + list(zip("ij",ijk[:2])))

	#----------------------------------------------------------------------
	@staticmethod
	def garc(g, x=None, y=None, z=None, i=None, j=None, k=None, **args):
		s = "g%d"%(g)
		if x is not None: s += ' '+CNC.fmt('x',x)
		if y is not None: s += ' '+CNC.fmt('y',y)
		if z is not None: s += ' '+CNC.fmt('z',z)
		if i is not None: s += ' '+CNC.fmt('i',i)
		if j is not None: s += ' '+CNC.fmt('j',j)
		if k is not None: s += ' '+CNC.fmt('k',k)
		for n,v in args.items():
			s += ' ' + CNC.fmt(n,v)
		return s
```

Note that the new additions for 6-axis also uses the explicit `for` loop instead of using a `zip` function.
```python
	#----------------------------------------------------------------------
	@staticmethod
	def _gotoABC(g, x=None, y=None, z=None, a=None, b=None, c=None, **args):
		s = "g%d"%(g)
		if x is not None: s += ' '+CNC.fmt('x',x)
		if y is not None: s += ' '+CNC.fmt('y',y)
		if z is not None: s += ' '+CNC.fmt('z',z)
		if a is not None: s += ' '+CNC.fmt('a',a)
		if b is not None: s += ' '+CNC.fmt('b',b)
		if c is not None: s += ' '+CNC.fmt('c',c)
		for n,v in args.items():
			s += ' ' + CNC.fmt(n,v)
		return s

	#----------------------------------------------------------------------
	@staticmethod
	def grapidABC(x=None, y=None, z=None, a=None, b=None, c=None, **args):
		return CNC._gotoABC(0,x,y,z,a,b,c,**args)

	#----------------------------------------------------------------------
	@staticmethod
	def glineABC(x=None, y=None, z=None, a=None, b=None, c=None, **args):
		return CNC._gotoABC(1,x,y,z,a,b,c,**args)
```

Actually, they can be combined into three functions total, making the code much simpler and easier to maintain. Each of the `*ABC` functions can replace the existing and omit the `*ABC` (keep the original names), since they are essentially the same thing.

Then, only `spiral.py` would need to have the functions renamed.
```python
		for g,x,y,z,r in zip(gP, xP,yP,zP, rP):
			if RotAxis == "A" :
				if g==0:
					block.append(CNC.grapidABC(x,y,z,r,CNC.vars["wb"],CNC.vars["wc"]))
					#sys.stdout.write("%s,%s,%s,%s,%s"%(g,x,y,z,r))
				else:
					block.append(CNC.glineABC(x,y,z,r,CNC.vars["wb"],CNC.vars["wc"]))
					#sys.stdout.write("%s,%s,%s,%s,%s"%(g,x,y,z,r))
			elif RotAxis == "B" :
				if g==0:
					block.append(CNC.grapidABC(x,y,z,CNC.vars["wa"],r,CNC.vars["wc"]))
				else:
					block.append(CNC.glineABC(x,y,z,CNC.vars["wa"],r,CNC.vars["wc"]))
			elif RotAxis == "C" :
				if g==0:
					block.append(CNC.grapidABC(x,y,z,CNC.vars["wa"],CNC.vars["wb"],r))
				else:
					block.append(CNC.glineABC(x,y,z,CNC.vars["wa"],CNC.vars["wb"],r))
```

Probably easier to do this now before others start coding for this.

tsmarks, 2020-05-16:
> 	# Note that we basically have six dupe (nine total) functions that do
> 	# the same thing, I would prefer the functions that are more explicit
> @tsmarks <https://github.com/tsmarks> What do you think about the
> proposal above?

That looks good to me.  I tried to avoid combining functions in the
beginning to reduce risk to the working XYZ code.  I have not had any
problems with the additions thus far, so I feel it is probably safe to
integrate the code into the XYZ functions.  I have been using the 4th axis
code over the last few weeks, but have not done any 3 axis work.  If no one
is having problems, I think it is good.

Tim

GitHubCaps, 2020-05-17:
@tsmarks I appreciate the feedback.

> I tried to avoid combining functions in the beginning to reduce risk to the working XYZ code.

Same here, always good to test, before proceeding.

I have a couple of questions for you. Any particular reason for not using `bool` here or okay with you to change?
```python
			("CutBoth",      "True,False",   "True", _("Cut in Both Directions")),
			("LiftPass",     "True,False",  "False", _("Lift before rotate")),
```
to:
```python
			("CutBoth",      "bool",   True, _("Cut in Both Directions")),
			("LiftPass",     "bool",  False, _("Lift before rotate")),
```
It would just change it to a `tickbox`, instead of a `listbox`. In my opinion it would be better, because the code doesn't check if it is `None` (incase user hasn't chosen either option).

Well, I said I have a couple of questions, but I will wait for the moment and put something together to present you with what I believe would help as far as using `Stock`, `Material`, `Endmill` databases etc. Wish we had our own forum for these discussions for improvement!

tsmarks, 2020-05-17:
I am not really sure why I used that code. But you are 100% correct.  A
bool would be much better.  I think, I started off with a different plan,
moved to the True/False and never updated it.  Sloppy by me.