vlachoudis/bCNC #1315: tkinter "TclError: named font does not already exist" at start up

tanah, 2019-12-08:
Just installed bCNC on Puppy Linux Bionic Beaver , with tkinter installed properly and the following error appears right after starting the program, crashing it. The window appears briefly without rendering the buttons in the left panel and then quickly crashes. I tried with the regular version from pip and now with the git version, without success (bCNC-0.9.14.301).

Please notice Im using python and not python2 since on puppy it refers to v2.7.

# python -m bCNC
new-config Utils <ConfigParser.ConfigParser instance at 0xb65d9e2c>
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
self.canvas.winfo_id(): 33554523
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/bCNC/__main__.py", line 2715, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/bCNC/__main__.py", line 2683, in main
    application = Application(tk)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/__main__.py", line 196, in __init__
    page = cls(self.ribbon, self)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/CNCRibbon.py", line 115, in __init__
    self.register()
  File "/usr/local/lib/python2.7/dist-packages/bCNC/EditorPage.py", line 609, in register
    (EditorFrame,))
  File "/usr/local/lib/python2.7/dist-packages/bCNC/CNCRibbon.py", line 134, in _register
    w = f(self.master._pageFrame, self.app)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/EditorPage.py", line 586, in __init__
    background=tkExtra.GLOBAL_CONTROL_BACKGROUND)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/CNCList.py", line 75, in __init__
    self.font      = tkFont.nametofont(self.cget("font"))
  File "/usr/lib/python2.7/lib-tk/tkFont.py", line 22, in nametofont
    return Font(name=name, exists=True)
  File "/usr/lib/python2.7/lib-tk/tkFont.py", line 85, in __init__
    raise Tkinter._tkinter.TclError, "named font %s does not already exist" % (self.name,)
_tkinter.TclError: named font -misc-fixed-medium-r-normal--14-130-75-75-c-70-*-* does not already exist

onekk, 2019-12-09:
Are you sure to have all the fonts installed it seems due to a lack of some fonts that Tkinter assume to have installed, or maybe in some bCNC.ini error or old user .bCNC in /home directory.

try to rename .bCNC in .bCNC.old in your name directory to force a clean generation of a new .bCNC.

Regards 

Carlo D.

tanah, 2019-12-09:
From what I've read it must be the lack of fonts but by the name given (-misc-fixed-medium-r-normal--14-130-75-75-c-70--) I don't know where it comes from, I hoped you could point me to a font library required by tkinter or maybe something else. 

I'll try that tonight and will let you know if it works.

Best regards

Harvie, 2019-12-09:
AFAIK Puppy Linux is rather minimalistic distribution, so they've probably stripped included fonts to bare minimum to save the diskspace. Maybe you can reconfigure bCNC to use some fonts that are available in your setup?

tanah, 2019-12-10:
I tried installing default font packages from ubuntu without success, reinstalling didn't work and there is no .bCNC directory in my whole computer, which I find very strange.

find / -name .bCNC
find: ‘/proc/12586’: No such file or directory
find: ‘/proc/12587’: No such file or directory
find: ‘/proc/12588’: No such file or directory
find: ‘/proc/12589’: No such file or directory
find: ‘/proc/12590’: No such file or directory

I checked bCNC.py and it seems it calls tkExtra for some fonts, although I don't have it installed and can't install it through pip, is that normal?

How do I reconfigure bCNC to use fonts of the system? Modifying the tkExtra lines? Because I also found the Helvetica font named in bCNC.py, installed it but didn't solve the problem.

onekk, 2019-12-10:
tkExtra is in /lib subdir of .bCNC is included in the bCNC "program tree".

To debug font name you have to install a font manager or use the fontconfig command line tools.

usually doing 
```
fc-list
```
will print the list of fonts installed, it could be a hint on how is missing.

Regards

Carlo D.