vlachoudis/bCNC #1318: Connection problem.

sfinexer, 2019-12-15:
I downloaded the latest version of b-CNC when searching for Home at the end of the program it says that there is no connection, however, the older version works correctly.

sfinexer, 2019-12-15:
[image]

sfinexer, 2019-12-15:
This version has no problems.
[bCNC-master_old.zip](https://github.com/vlachoudis/bCNC/files/3965327/bCNC-master_old.zip)

sfinexer, 2019-12-15:
Old version:
[image]

New Version:
[image]

sfinexer, 2019-12-16:
> sfinexer@sfinexer:~$ locale
LANG=ru_RU.UTF-8
LANGUAGE=
LC_CTYPE="ru_RU.UTF-8"
LC_NUMERIC="ru_RU.UTF-8"
LC_TIME="ru_RU.UTF-8"
LC_COLLATE="ru_RU.UTF-8"
LC_MONETARY="ru_RU.UTF-8"
LC_MESSAGES="ru_RU.UTF-8"
LC_PAPER="ru_RU.UTF-8"
LC_NAME="ru_RU.UTF-8"
LC_ADDRESS="ru_RU.UTF-8"
LC_TELEPHONE="ru_RU.UTF-8"
LC_MEASUREMENT="ru_RU.UTF-8"
LC_IDENTIFICATION="ru_RU.UTF-8"
LC_ALL=

MARIOBASZ, 2019-12-16:
I have a similar problem: (I wrote it last week, I don't see it in the list)
I downloaded the program to an old PC with windows xp;
Connect, grbl respond and move, read response on terminal screen, but bCNC appears disconnected, does not show machine or work coordinates, and on the editing screen, the block always appears with the box checked enabled unchecked, it does change color (highlighted or smoothed) as marked / unmarked

sfinexer, 2019-12-16:
> I have a similar problem: (I wrote it last week, I don't see it in the list)
> I downloaded the program to an old PC with windows xp;
> Connect, grbl respond and move correctly, read response on terminal screen, but bCNC appears disconnected, does not show machine or work coordinates, and on the editing screen, the block always appears with the box checked enabled unchecked, it does change color (highlighted or smoothed) as marked / unmarked

$10 = ???
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#10---status-report-mask

sfinexer, 2019-12-16:
My machine stops responding!

MARIOBASZ, 2019-12-16:
@sfinexer , thanks, but with the same cnc connected to my bCNC notebook it works fine, I rule out grbl configuration problem.

sfinexer, 2019-12-16:
I FIND WHERE AN ERROR !!!
The error appeared after making the changes:
https://github.com/vlachoudis/bCNC/commit/4c20e751566679e1a510d687b25200c0bd23cd47

Harvie, 2019-12-16:
> The error appeared after making the changes:
> [4c20e75](https://github.com/vlachoudis/bCNC/commit/4c20e751566679e1a510d687b25200c0bd23cd47)

What python version do you have? Works for me in both 3.8.0 and 2.7.17

sfinexer, 2019-12-16:
sfinexer@sfinexer:~$ python --version
Python 2.7.17

sfinexer, 2019-12-18:
@Harvie,
I think you understand that the error appears ONLY when searching for Home?
When executing a different code, there is no error!

rjordans, 2020-04-22:
I have the same problem here, both in pyton 2.7.17 and python 3.6.9 (Ubuntu 18.04).  Checking out the version before the commit mentioned above indeed fixes the problem (python2 only though) for me.

The problem occurs here as soon as I connect to the machine (one of these cheapo Chinese ones).

lbattraw, 2020-05-16:
This error can be fixed by editing the line that was changed, from this at line 766 of bCNC/Sender.py:

line = str(self.serial.readline().decode()).strip()

to this:

line = str(self.serial.readline().decode("ascii", "ignore")).strip()

This fixes it for me, I'm running Python 3.7.5.  

BTW I'm trying to use bCNC with the Grbl_Esp32, and something about bCNC really kills it and it gets stuck in a reboot loop.  It's too bad since Grbl_Esp32 is just amazing, everything can be run from a web page hosted on it but it doesn't offer the same tight control a serial connection provides, along with all the tools in bCNC.  I figure bCNC is trying to send something right as it's booting which crashes it, rinse and repeat:

Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
[MSG:Grbl_ESP32 Ver 1.2a Date 20200428]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:ESP32_V4]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Client Started]
[MSG:Connecting TP-LINK_9D38]
[MSG:Connecting.]
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e051e  PS      : 0x00060e30  A0      : 0x800e07b0  A1      : 0x3ffd2f40
A2      : 0x00000000  A3      : 0x3ffc5a04  A4      : 0x3ffc855c  A5      : 0x00000009
A6      : 0x0000007c  A7      : 0x00000001  A8      : 0x800e0519  A9      : 0x3ffd2f20
A10     : 0x00000000  A11     : 0x3ffd2f74  A12     : 0x000000a0  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff
Backtrace: 0x400e051e:0x3ffd2f40 0x400e07ad:0x3ffd30b0 0x400e09c2:0x3ffd30d0 0x400919ed:0x3ffd30f0
Rebooting...

Wish I could figure out how to configure bCNC to leave it alone until it's done booting, or what command could be crashing it.  It crashes while Grbl_ESP32 is connecting to my wifi AP...

MARIOBASZ, 2020-06-10:
Old PC.
Working well, but when restarting, same problem discussed. Python 2.7 is loaded.
It is listed as not connected.
If I move axes, it does it in jog mode.
Buttons to set values to 0: they work.
Make file run: ok.
Stop button: ok
Spindle button and speed: does not work.
In another version of bcnc it continuously receives position status.
Uncomfortable problem, hopefully someone can fix it.

cuchaz, 2021-03-27:
I ran into a very similar problem and came here to say that @lbattraw 's solution worked for me!

Ignoring the errors from the ASCII/UTF-8 decoding (ie, that second "ignore" argument) seems to work around my problem. Apparently my GRBL controller occasionally sends non-character bytes down the pipe? The job finished just fine though and I didn't notice any bad side-effects from ignoring the character decoding errors. Although there was one "garbage received" message or something similar that flew past in bCNC's terminal display. I couldn't copy it though because the window auto-scrolled until the message got pushed off the top of the buffer, but I think the message mentioned something about a position. Maybe? It was a blur.

Harvie, 2021-03-27:
> This error can be fixed by editing the line that was changed, from this at line 766 of bCNC/Sender.py:
> line = str(self.serial.readline().decode()).strip()
> to this:
> line = str(self.serial.readline().decode("ascii", "ignore")).strip()
> This fixes it for me, I'm running Python 3.7.5.

Interresting, might use this fix. Maybe even in other places like g-code parsing?

> Apparently my GRBL controller occasionally sends non-character bytes down the pipe?

Not sure, you might check the protocol specifications.
However if there is electromagnetic interference (aka noise) somewhere on the line, it is highly probable that some random bytes might be received by bCNC.

cuchaz, 2021-03-27:
I managed to snag one of those garbage console messages as it zoomed by:
```
Garbage receive MPos: <Run|MPos:-356.713,-366.88ir	0,50|FS:120,10000>
```
No idea if that's helpful or not, but there you go.

dzervas, 2021-12-08:
I wanna chip in and say that I have the same problem in python 3.9.

I think that the problem is that bCNC resets the controller altogether. In Arduino it might make sense but in ESP32 it doesn't, as we have a state as we go, wifi connection, etc.
After the ESP32 is reset, it sends some gibberish (after applying your patch @lbattraw which saved me from throwing my 2 years of work out the window <3 ) which I guess could be baudrate negotiation.

All in all, by making reset on connect optional this problem is gone (while I understand that it's FluidNC's and Grbl-ESP32's problems actually) and we don't lose our wifi access as well

dzervas, 2021-12-08:
[image]

dzervas, 2021-12-08:
Yeap, checks out, I commented out the two `setDTR(0)` blocks in Sender.py in the `open` function and it works like a charm! also the sleeps are not needed and the connection is super fast :)

maduka-ariyasiri, 2022-07-11:
I have submitted a PR. I would appreciate if you could review the changes.