vlachoudis/bCNC #1320: bCNC cannot launch on debian with french locale

delattrepa, 2019-12-20:
Using bCNC, I have found some bug related to french locale on my debian/KDE. The bug is not present when I change the regional settings to the default settings. 

Here is the bug :
python2 -m bCNC
new-config Utils <ConfigParser.ConfigParser instance at 0x7f9b2f2e4440>
================================================================================
WARNING: bCNC has been recently ported to support both python v2.x and v3.x
Most things seem to work reasonably well in both python versions.
Please report any issues to: https://github.com/vlachoudis/bCNC/issues
================================================================================
self.canvas.winfo_id(): 123732059
/usr/local/lib/python2.7/dist-packages/bCNC/ProbePage.py:1876: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  CNC.toolWaitAfterProbe = int(TOOL_WAIT.index(self.toolWait.get()))
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/bCNC/__main__.py", line 2715, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/bCNC/__main__.py", line 2683, in main
    application = Application(tk)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/__main__.py", line 196, in __init__
    page = cls(self.ribbon, self)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/CNCRibbon.py", line 115, in __init__
    self.register()
  File "/usr/local/lib/python2.7/dist-packages/bCNC/ProbePage.py", line 1987, in register
    (ProbeCommonFrame, ProbeFrame, AutolevelFrame, CameraFrame, ToolFrame))
  File "/usr/local/lib/python2.7/dist-packages/bCNC/CNCRibbon.py", line 134, in _register
    w = f(self.master._pageFrame, self.app)
  File "/usr/local/lib/python2.7/dist-packages/bCNC/ProbePage.py", line 1667, in __init__
    self.toolWait.set(TOOL_WAIT[1])
  File "/usr/local/lib/python2.7/dist-packages/bCNC/lib/tkExtra.py", line 3094, in set
    self.invoke()
  File "/usr/local/lib/python2.7/dist-packages/bCNC/lib/tkExtra.py", line 3046, in invoke
    self.command()
  File "/usr/local/lib/python2.7/dist-packages/bCNC/ProbePage.py", line 1876, in waitChange
    CNC.toolWaitAfterProbe = int(TOOL_WAIT.index(self.toolWait.get()))
ValueError: u'AVANT & APR\xc8S sondage' is not in list

I gess it is a problem with Unicode and the special character È (uppercase E with grave accent) in the following file :

https://github.com/vlachoudis/bCNC/blob/master/bCNC/locale/fr/LC_MESSAGES/bCNC.po
#: bCNC/ProbePage.py:44
msgid "BEFORE & AFTER probing"
msgstr "AVANT & APRÈS sondage"

gnurc, 2019-12-27:
Hello

I have the same proble with Spanish language.

BR

ted0xff, 2019-12-28:
Hello,

you should try to update your bCNC copy (pip2 install --upgrade bCNC).
I have a debian in French and no problem with locale
[image]

ted0xff, 2019-12-28:
So currently, you have two solutions:
Stay in python2 (and never again update):
```
pip2 install --upgrade bCNC==0.9.14.62
python2 -m bCNC
```

Or go to python3 (both can co-habitate): 
```
pip3 install --upgrade bCNC
python3 -m bCNC
```

This time I tested both installations

delattrepa, 2019-12-28:
The upgrade in python2 with version 0.9.14.62 works as well as the new installation in python3.

Thanks.

closed 2019-12-28 by delattrepa