vlachoudis/bCNC #1326: Optimize function

dodolbysuround, 2019-12-26:
Optimize funtion seems to fail.
[image]

DodoLaSaumure, 2019-12-29:
Hello
you can upgrade to the lastest version, where this issue is now corrected
`pip3 install --upgrade git+https://github.com/vlachoudis/bCNC`

dodolbysuround, 2019-12-29:
Thank you so much! this is perfect.
I wish I could help debogging by myself but although I'm trying my hand to python, my level is still ridiculous...

Harvie, 2019-12-30:
> I wish I could help debogging by myself but although I'm trying my hand to python, my level is still ridiculous...

haha. i've learned python by improving bCNC. didn't knew anything about python before :-)

closed 2019-12-30 by Harvie