vlachoudis/bCNC #1327: copy / cut doesn't work

dodolbysuround, 2019-12-27:
Hi, it seems copy / cut functions doewm't work properly
that's chat I get:
[image]

the abtou bottun also generates an error:
[image]

Harvie, 2019-12-29:
Try latest version pls, @DodoLaSaumure seems to have fixed that...

Harvie, 2020-12-28:
Reportedly fixed in this PR. Please test and close this issue if everything is OK:
https://github.com/vlachoudis/bCNC/pull/1405#issuecomment-751580531

dodolbysuround, 2021-01-07:
Thank you so much for all your work! issue solved!!

closed 2021-01-07 by dodolbysuround