vlachoudis/bCNC #1331: Problem comm Ethernet smoothie

ludo10SA2I, 2019-12-29:
Hello,
I am working on a CNC milling. I researched a little all the software allowing me to control my machine. I own a smoothieboard and I first wanted to use BCNC. The problem is that when I connect via Ethernet. He tells me that I am connected. But the interface does not seem to be connected because I never receive the position of my axes, moreover when I try to make movements it does not move all the time and randomly.
I tried to connect to usb but the driver does not install correctly on my pc.
So I turned to SMOOPI.
Everything works except that I do not get the coordinates of the axes either ... In short I don't understand anything but maybe you could give me a track?
FYI, bcnc runs windows 7
The smoothie firmware is:
edge-2b84ffc Jul 3 2019 21:32:06
GRBL_MODE True
Thank you in advance.
Ludovic Taillez

Harvie, 2019-12-29:
Can you please launch bCNC from commandline? It will print some debugging messages to the black commandline window should something go wrong...

ludo10SA2I, 2020-01-02:
Hello,
first of all thank you for your help
it is written in the window:

self.canvas.w_infoID () 394110
Using old style comports ()!
Using old style comports ()!
Using old style comports ()!
My version of bcnc is version 0.9.14.
I'm on Win 7 64bits
My PIP version is 19.3.1
[https://drive.google.com/open?id=0B2Sm_hrIZ3tiOG5CaHgtS2xzTzhPb2d1S0dyWGR3eEhJWWhV](url)