vlachoudis/bCNC #1335: bCNC Remaining in idle state

DodoLaSaumure, 2020-01-04:
I do not know why but with the following gCode, controller stays in idle state at z=-12
My stock is 20mm thick

(Block-name: Header)
(Block-expand: 0)
(Block-enable: 1)
M3 S12000
G4 P3
G0 Z10
(Block-name: Drill)
(Block-expand: 1)
(Block-enable: 1)
g0 x0 y0
g0 z10
g0 z0.6
g1 z0
(entered)
g1 z-3 f50
g0 z0.6
g4 p0.2
g0 z-2.4
g1 z-3
g1 z-6 f50
g0 z0.6
g4 p0.2
g0 z-5.4
g1 z-6
g1 z-9 f50
g0 z0.6
g4 p0.2
g0 z-8.4
g1 z-9
g1 z-12 f50
g0 z0.6
g4 p0.2
g0 z-11.4
g1 z-12
g1 z-15 f50
g0 z0.6
g4 p0.2
g0 z-14.4
g1 z-15
g1 z-18 f50
g0 z0.6
g4 p0.2
g0 z-17.4
g1 z-18
g1 z-20 f50
g0 z0.6
g4 p0.2
(exiting)
g0 z10
(Block-name: Footer)
(Block-expand: 0)
(Block-enable: 1)
M5
G0 Z40
G0 X0 Y0

MARIOBASZ, 2020-01-04:
See if it doesn't throw error 22:
it seems that the value F for g1 is not set

DodoLaSaumure, 2020-01-04:
Thank you Mario, setting the feed solves the problem.
But I do not see why it waits Z=-12.
Thanks
Pierre

closed 2020-01-04 by DodoLaSaumure

MARIOBASZ, 2020-01-04:
If I am not wrong, the error occurred on the line
g1z0
(line 7)
error 22
but it does not stop immediately.
The why, I don't know, experience of making mistakes often

DodoLaSaumure, 2020-01-04:
@MARIOBASZ 
Thank you Mario, I agree with you, because setting the feed on line 7 solves the problem
I close the issue since it is solved.
Thanks 
Best regards
Pierre